moveit/moveit2 #663: The complete state of the robot is not yet known.  Missing virtual_joint

Maxcode19, 2021-09-02:
Hello. I have just built moveit2 for Foxy and I am trying to start the demo:

`ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`

The demo seems to start correctly, but, after a while, the following warning comes out:

_`[run_moveit_cpp-4] [WARN] [1630594527.651254089] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint`_

In Rviz, this warning corresponds to an incorrect visualization of the arm. The arm "oscillates" between the correct pose and an incorrect one. An example of the behaviour is the given by a continuous and fast alternation between the following two poses:

[image]

[image]

Unfortunately, this problem happens also with other demos, like the one from moveit2_tutorials:

`ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true`

I have read about a similar problem [here](https://github.com/ros-planning/moveit2/issues/251) but, as far as I got, it has been solved. Do you have any suggestion to solve this inconvenient?

Thanks in advance

vatanaksoytezer, 2021-09-02:
Hello @Maxcode19 are you using binaries or source build? Which distribution? If you are using binaries you may have came across with https://github.com/ros-planning/moveit2_tutorials/issues/111#issuecomment-911858695.

Maxcode19, 2021-09-03:
Hello @vatanaksoytezer! Thank you very much for your quick reply. I start reporting a good news. Today the wrong behaviour that I experienced yesterday has disappeared. I have tried several time to reproduce it, but, luckily, it has not taken place again.

Anyway, just to describe in more details my situation, I have installed Moveit2 from source for Foxy on Ubuntu 18.04 (I know that Moveit2 is not intended to work with it, but Ubuntu 18.04 is a constraint I have). Actually, I have installed 2 different versions of Moveit2, one cloned yesterday and one cloned some months ago. Yesterday, both of them had the issue of the missing joint, today none of them. The only difference between the two versions is that the new one does not always start correctly (it does not always trigger everything) and leads to a slower motion of the arm with respect to the older one (but there could be some parameters set in a different way).

So, for now, I think that this issue can be closed. Thanks for the help.

closed 2021-09-03 by Maxcode19