moveit/moveit2 #677: Plan but do not Execute on Galactic version of Moveit2

ggiulian, 2021-09-13:
### Description

When I execute the plan command on Rviz, the robot plans the trajectory but when it is time to execute it fails. It should be noted that the same robot on ROS2 FOXY and MOVEIT2 for Foxy works,  i.e. planning is followed by perfect execution of the trajectory. P.S i am using a personal robot

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source or Binary build?
 Build by source

The steps to reproduce the issue are:

1. Launch the move group and display it with rviz
2. Set the "Goal State" to random
3. Press Plan button
4. Press Execute button
5. Execution Failed

### Expected behaviour
Execution of the trajectory after planning

### Actual behaviour
Trajectory planning, but not execution

JafarAbdi, 2021-09-13:
@ggiulian Could you please share the logs .?

ggiulian, 2021-09-13:
[move_group-4] [INFO] [1631552348.270267036] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-4] [INFO] [1631552348.270743756] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-4] [INFO] [1631552348.270835345] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-4] [ERROR] [1631552348.270881522] [moveit_ros.trajectory_execution_manager]: Unable to identify any set of controllers that can actuate the specified joints: [ joint0 joint1 joint2 joint3 ]
[move_group-4] [ERROR] [1631552348.270905235] [moveit_ros.trajectory_execution_manager]: Known controllers and their joints:
[move_group-4] 

[rviz2-1] [INFO] [1631552348.672194229] [move_group_interface]: Execute request accepted
[ros2_control_node-5] [INFO] [1631552348.727963061] [RoboertoSystemPositionHardware]: Writing...
[ros2_control_node-5] [INFO] [1631552348.728053408] [RoboertoSystemPositionHardware]: Writing...
[ros2_control_node-5] [INFO] [1631552348.728069084] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728135324] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728152568] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728164656] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728178657] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728190763] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728203112] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728214983] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728227060] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728239161] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728251159] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728263002] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728275121] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728287107] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728299702] [RoboertoSystemPositionHardware]: Got command 0.00000!
[ros2_control_node-5] [INFO] [1631552348.728311668] [RoboertoSystemPositionHardware]: Got command 0.00000!
[rviz2-1] [INFO] [1631552348.773264968] [move_group_interface]: Execute request aborted
[rviz2-1] [ERROR] [1631552348.873752867] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached

JafarAbdi, 2021-09-13:
The problem is right here

> [move_group-4] [INFO] [1631552348.270835345] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-4] [ERROR] [1631552348.270881522] [moveit_ros.trajectory_execution_manager]: Unable to identify any set of controllers that can actuate the specified joints: [ joint0 joint1 joint2 joint3 ]

Can you make sure that you started the controllers .? also you mind sharing the moveit simple controller's yaml file or the moveit configs package .?

ggiulian, 2021-09-14:
Yes, this is what I write:
`# Trajectory Execution Functionality
    moveit_simple_controllers_yaml = load_yaml(
        "moveit_resources_roboerto_config", "config/roboerto_controllers.yaml"
    )
    moveit_controllers = {
        "moveit_simple_controller_manager": moveit_simple_controllers_yaml,
        "moveit_controller_manager": "moveit_simple_controller_manager/MoveItSimpleControllerManager",
    }

    trajectory_execution = {
        "moveit_manage_controllers": True,
        "trajectory_execution.allowed_execution_duration_scaling": 1.2,
        "trajectory_execution.allowed_goal_duration_margin": 0.5,
        "trajectory_execution.allowed_start_tolerance": 0.01,
    }

    planning_scene_monitor_parameters = {
        "publish_planning_scene": True,
        "publish_geometry_updates": True,
        "publish_state_updates": True,
        "publish_transforms_updates": True,
    }
`

JafarAbdi, 2021-09-14:
I need the content of this file `roboerto_controllers.yaml`

ggiulian, 2021-09-14:
This is the content of `roboerto_controllers.yaml`

```
controller_names:
  - roboerto_arm_controller

roboerto_arm_controller:
  action_ns: follow_joint_trajectory
  type: FollowJointTrajectory
  default: true
  joints:
    - joint0
    - joint1
    - joint2
    - joint3
```

JafarAbdi, 2021-09-14:
This probably means that you're not spawning the controllers, you could do it similar to https://github.com/ros-planning/moveit2/blob/main/moveit_demo_nodes/run_move_group/launch/run_move_group.launch.py#L170-L182, also I do need the whole logs for me to see what's actually happening

ggiulian, 2021-09-14:
This is what I did to spawn the controller:

```
 # Load controllers
    load_controllers = []
    for controller in ["roboerto_arm_controller", "joint_state_broadcaster"]:
        load_controllers += [
            ExecuteProcess(
                cmd=["ros2 run controller_manager spawner.py {}".format(controller)],
                shell=True,
                output="screen",
            )
        ]
```

So, I think it is the same of the link.
In the next comment I'll post the whole logs

ggiulian, 2021-09-14:
[launch.log](https://github.com/ros-planning/moveit2/files/7163131/launch.log)

JafarAbdi, 2021-09-14:
There are two errors

```
1631632552.2915051 [ERROR] [ros2 run controller_manager spawner.py roboerto_arm_controller-6]: process has died [pid 6591, exit code 1, cmd 'ros2 run controller_manager spawner.py roboerto_arm_controller'].
1631632552.2956188 [ERROR] [ros2 run controller_manager spawner.py joint_state_broadcaster-7]: process has died [pid 6593, exit code 1, cmd 'ros2 run controller_manager spawner.py joint_state_broadcaster'].
```

Do you know why they're dying .? what happens when you call them manually .? `ros2 run controller_manager spawner.py roboerto_arm_controller`

ggiulian, 2021-09-14:
I don't know why they are dying, but this is what happen when I call them manually:
```
ggiuliano@Vivobook:~$ ros2 run controller_manager spawner.py roboerto_arm_controller
/home/ggiuliano/roboerto_g_moveit2_ws/install/controller_manager/lib/controller_manager/spawner.py:186: DeprecationWarning: 'spawner.py' is deprecated, please use 'spawner' (without .py extension)
  warnings.warn(
[INFO] [1631633719.658323258] [spawner_roboerto_arm_controller]: Loaded roboerto_arm_controller
[INFO] [1631633719.662327624] [spawner_roboerto_arm_controller]: Failed to configure controller
```

JafarAbdi, 2021-09-14:
Sorry I have no idea why the spawner is failing, if you can share your moveit config I might be able to try to reproduce it locally

Closing since it's not related to moveit

closed 2021-09-14 by JafarAbdi

ggiulian, 2021-09-14:
@JafarAbdi here the GitHub link to my moveit config of Roboerto:
https://github.com/ggiulian/Roboerto_moveit2.git

JafarAbdi, 2021-09-14:
> @JafarAbdi here the GitHub link to my moveit config of Roboerto:
> https://github.com/ggiulian/Roboerto_moveit2.git

I'm getting 404

ggiulian, 2021-09-14:
@JafarAbdi  try again https://github.com/ggiulian/Roboerto_moveit2 , now  it should work.
Anyway, I moved the discussion here: https://github.com/ggiulian/Roboerto_moveit2/issues/1