moveit/moveit2 #678: Hardware_interface Failed

kashifzr, 2021-09-14:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source Build
* If binary, which release version?
* If source, which branch?

### Steps to reproduce

colcon build --event-handlers desktop_notification- status- --cmake-args

### Expected behaviour
successful build 

### Actual behaviour
hardware_inter face failed 

### Backtrace or Console output
<pre>colcon build --event-handlers desktop_notification- status- --cmake-args
Starting &gt;&gt;&gt; geometric_shapes
Starting &gt;&gt;&gt; moveit_msgs
Starting &gt;&gt;&gt; srdfdom
Starting &gt;&gt;&gt; ros2_control_test_assets
Finished &lt;&lt;&lt; ros2_control_test_assets [7.62s]
Starting &gt;&gt;&gt; hardware_interface
--- stderr: hardware_interface
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::ClassLoader::isLibraryLoaded()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::package::getPath(std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::initialize()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, ros::console::levels::Level)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::path::filename() const&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::MultiLibraryClassLoader::getAllAvailableClassLoaders()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&amp;, boost::system::error_code*)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::g_initialized&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::MultiLibraryClassLoader::getRegisteredLibraries[abi:cxx11]()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::package::getPlugins(std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, std::vector&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt;, std::allocator&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; &gt; &gt;&amp;, bool)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&amp;)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::path::parent_path() const&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::isOwnedBy(class_loader::ClassLoader const*)&apos;
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_component_interfaces.dir/build.make:141: test_component_interfaces] Error 1
make[1]: *** [CMakeFiles/Makefile2:438: CMakeFiles/test_component_interfaces.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::ClassLoader::isLibraryLoaded()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::package::getPath(std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::initialize()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, ros::console::levels::Level)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::path::filename() const&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::MultiLibraryClassLoader::getAllAvailableClassLoaders()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&amp;, boost::system::error_code*)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::g_initialized&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::MultiLibraryClassLoader::getRegisteredLibraries[abi:cxx11]()&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `ros::package::getPlugins(std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, std::vector&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt;, std::allocator&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; &gt; &gt;&amp;, bool)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&amp;)&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `boost::filesystem::path::parent_path() const&apos;
/usr/bin/ld: libhardware_interface.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::isOwnedBy(class_loader::ClassLoader const*)&apos;
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_generic_system.dir/build.make:141: test_generic_system] Error 1
make[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/test_generic_system.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   &lt;&lt;&lt; hardware_interface [21.5s, exited with code 2]
Aborted  &lt;&lt;&lt; geometric_shapes [29.3s]
Aborted  &lt;&lt;&lt; srdfdom [29.2s]
Aborted  &lt;&lt;&lt; moveit_msgs [29.9s]

Summary: 1 package finished [30.5s]
  1 package failed: hardware_interface
  3 packages aborted: geometric_shapes moveit_msgs srdfdom
  2 packages had stderr output: geometric_shapes hardware_interface
  49 packages not processed
</pre>

No idea why this colcon build failed, Getting Worst Experence with Ubuntu 20.04

AndyZe, 2021-09-14:
It looks like this is a ros2_control issue, not a MoveIt issue. But anyway, I'd start by checking that you have Boost installed.

Are you using Galactic debians (installed by `apt install ros-galactic-...`), Rolling debians, or are you building from source?

AndyZe, 2021-09-14:
Basically, update the information that is asked for at the top of the issue template. ROS Distro, etc

kashifzr, 2021-09-14:
I am using Using Ros Noetic and Ros Foxy. 
Yes I am building from Source 
I know some how its related to ros2_control but I need to install ros2_control to work with moveit2

AndyZe, 2021-09-14:
OK, thanks. Please confirm that you are only sourcing Noetic or Foxy, not both at the same time. You should have something like this in your .bashrc. Note that Noetic is commented.

# source /opt/ros/noetic/setup.bash
source /opt/ros/foxy/setup.bash

AndyZe, 2021-09-14:
The easiest way fix might be to re-install Ubuntu then carefully follow the instructions here:

https://moveit.ros.org/install-moveit2/source/

kashifzr, 2021-09-14:
When I am building Moveit2 from Source I source with ROS 2 Foxy, with source command "source /opt/ros/foxy/setup.bash"

kashifzr, 2021-09-14:
I will try to reinstall Ubuntu Since I updated it from Ubuntu 18 it will creating issues.
Thanks

JafarAbdi, 2021-09-14:
I see you also opened an issue in [moveit](https://github.com/ros-planning/moveit/issues/2874), which mean there's a big chance that some of your env-variables are missed up

Could you please share the value of the following env-variables ?

```
LD_LIBRARY_PATH
CMAKE_PREFIX_PATH
CMAKE_MODULE_PATH
AMENT_PREFIX_PATH
AMENT_CURRENT_PREFIX
COLCON_PREFIX_PATH
```

kashifzr, 2021-09-14:
How to link with Building Moveit from Source when they may or not used in Building. Can you share the command where these are used. 
As Far As I know they may have values when Build will be successful, In my case Build is not working

If want to know exactly these variables then i will share when after installations

JafarAbdi, 2021-09-28:
This's related to not having a clean env-variable, closing for now, please feel free to re-open it if you still have the problem

closed 2021-09-28 by JafarAbdi

jainishmehta, 2023-05-16:
HI, I have the same issue. I am using Ubuntu 20.04 focal and have foxy
[image]
I git cloned from https://github.com/ros-controls/ros2_control/ and then did colcon build to get the package on my ros2, however it doesn't work.