moveit/moveit2 #681: moveit_core Build Failure

nathankaleida, 2021-09-16:
### Description

moveit_core build fails at ruckig dependency

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
Binary build
`ros-galactic-desktop/focal,now 0.9.3-2focal.20210820.074533 amd64 [installed]`
`ros-galactic-ruckig/focal,now 0.3.3-3focal.20210810.134905 amd64 [installed]`

### Steps to reproduce
Encountered error following UR Driver installation procedure: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver

### Expected behaviour
Builds without errors

### Actual behaviour
Build fails

```--- stderr: moveit_core                                           
/home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp: In static member function ‘static bool trajectory_processing::RuckigSmoothing::applySmoothing(robot_trajectory::RobotTrajectory&, double, double)’:
/home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:79:47: error: no matching function for call to ‘ruckig::Ruckig<0>::Ruckig(<brace-enclosed initializer list>)’
   79 |   ruckig::Ruckig<0> ruckig{ num_dof, timestep };
      |                                               ^
In file included from /home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h:41,
                 from /home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:40:
/opt/ros/galactic/include/ruckig/ruckig.hpp:52:14: note: candidate: ‘ruckig::Ruckig<DOFs, throw_error, return_error_at_maximal_duration>::Ruckig(double) [with long unsigned int DOFs = 0; bool throw_error = false; bool return_error_at_maximal_duration = true]’
   52 |     explicit Ruckig(double delta_time): delta_time(delta_time) { }
      |              ^~~~~~
/opt/ros/galactic/include/ruckig/ruckig.hpp:52:14: note:   candidate expects 1 argument, 2 provided
/opt/ros/galactic/include/ruckig/ruckig.hpp:51:14: note: candidate: ‘ruckig::Ruckig<DOFs, throw_error, return_error_at_maximal_duration>::Ruckig() [with long unsigned int DOFs = 0; bool throw_error = false; bool return_error_at_maximal_duration = true]’
   51 |     explicit Ruckig() { }
      |              ^~~~~~
/opt/ros/galactic/include/ruckig/ruckig.hpp:51:14: note:   candidate expects 0 arguments, 2 provided
In file included from /home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h:41,
                 from /home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:40:
/opt/ros/galactic/include/ruckig/ruckig.hpp:22:7: note: candidate: ‘constexpr ruckig::Ruckig<0>::Ruckig(const ruckig::Ruckig<0>&)’
   22 | class Ruckig {
      |       ^~~~~~
/opt/ros/galactic/include/ruckig/ruckig.hpp:22:7: note:   candidate expects 1 argument, 2 provided
/opt/ros/galactic/include/ruckig/ruckig.hpp:22:7: note: candidate: ‘constexpr ruckig::Ruckig<0>::Ruckig(ruckig::Ruckig<0>&&)’
/opt/ros/galactic/include/ruckig/ruckig.hpp:22:7: note:   candidate expects 1 argument, 2 provided
/home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:80:51: error: no matching function for call to ‘ruckig::InputParameter<0>::InputParameter(<brace-enclosed initializer list>)’
   80 |   ruckig::InputParameter<0> ruckig_input{ num_dof };
      |                                                   ^
In file included from /opt/ros/galactic/include/ruckig/ruckig.hpp:13,
                 from /home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h:41,
                 from /home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:40:
/opt/ros/galactic/include/ruckig/input_parameter.hpp:70:5: note: candidate: ‘ruckig::InputParameter<DOFs>::InputParameter() [with long unsigned int DOFs = 0]’
   70 |     InputParameter() {
      |     ^~~~~~~~~~~~~~
/opt/ros/galactic/include/ruckig/input_parameter.hpp:70:5: note:   candidate expects 0 arguments, 1 provided
/opt/ros/galactic/include/ruckig/input_parameter.hpp:43:7: note: candidate: ‘constexpr ruckig::InputParameter<0>::InputParameter(const ruckig::InputParameter<0>&)’
   43 | class InputParameter {
      |       ^~~~~~~~~~~~~~
/opt/ros/galactic/include/ruckig/input_parameter.hpp:43:7: note:   no known conversion for argument 1 from ‘const size_t’ {aka ‘const long unsigned int’} to ‘const ruckig::InputParameter<0>&’
/opt/ros/galactic/include/ruckig/input_parameter.hpp:43:7: note: candidate: ‘constexpr ruckig::InputParameter<0>::InputParameter(ruckig::InputParameter<0>&&)’
/opt/ros/galactic/include/ruckig/input_parameter.hpp:43:7: note:   no known conversion for argument 1 from ‘const size_t’ {aka ‘const long unsigned int’} to ‘ruckig::InputParameter<0>&&’
/home/nkaleida/ur_ws/src/moveit2/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:81:53: error: initializer for ‘std::__array_traits<double, 0>::_Type’ must be brace-enclosed
   81 |   ruckig::OutputParameter<0> ruckig_output{ num_dof };
      |                                                     ^
make[2]: *** [trajectory_processing/CMakeFiles/moveit_trajectory_processing.dir/build.make:89: trajectory_processing/CMakeFiles/moveit_trajectory_processing.dir/src/ruckig_traj_smoothing.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1931: trajectory_processing/CMakeFiles/moveit_trajectory_processing.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [3min 57s, exited with code 2]

vatanaksoytezer, 2021-09-16:
@nathankaleida thanks for the query! Ruckig was just merged with #571. Can you try install dependencies to pull rucking as well by issuing a rosdep install command: 

ie. you can issue the following command after you cd into `ws_moveit/src`:

```
rosdep install -r --from-paths . --ignore-src --rosdistro galactic -y
```

JafarAbdi, 2021-09-28:
This should be fixed now

closed 2021-09-28 by JafarAbdi