moveit/moveit2 #691: Removing `source` job from ROS 2 buildfarm

gavanderhoorn, 2021-09-21:
Would it be an idea to remove the `source` job from the ROS 2 buildfarm for the time being?

We already have CI here via GHA and the builds have been failing / unstable for at least 4 months now if not longer.

They're either timing out or failing due to something which doesn't come up on the GHA builds.

In addition to avoiding email spam, we would also save OR a couple of compute-dollars.

gavanderhoorn, 2021-09-21:
My plan is to submit PRs removing those entries at the end of this week if there are no comments here.

gavanderhoorn, 2021-09-24:
https://github.com/ros/rosdistro/pull/30861

henningkayser, 2021-09-28:
Hi @gavanderhoorn, I agree that these jobs are probably redundant. Rolling doesn't fail though, the others have unmet dependencies or breaking changes.

gavanderhoorn, 2021-09-28:
The rolling job is only back to stable since last week. That was after I submitted my PR.

If we feel there is value in having a duplicate job for rolling running on the farm, it could be taken out of the removal-PR.

closed 2021-09-30 by gavanderhoorn