moveit/moveit2 #708: moveit_ros/perception BUG

YongLD, 2021-09-27:
### Description

no known conversion for argument 1 from occupancy_map_monitor::OccMapTreePtr to const OccMapTreePtr&

### Your environment
* ROS Distro: [Galactic]
* OS Version: e.g. Ubuntu 20.04
* Source build
* branch: main

### Steps to reproduce
step of the moveit-base from "https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver"

### Expected behaviour
All packages built

### Actual behaviour
moveit_ros build bug

### Backtrace or Console output

> Starting >>> moveit_ros_perception
[Processing: moveit_kinematics, moveit_planners_ompl, moveit_ros_perception, moveit_ros_robot_interaction, moveit_ros_warehouse]
--- stderr: moveit_ros_perception
/home/wislab/ros_robot/src/moveit2/moveit_ros/perception/depth_image_octomap_updater/src/depth_image_octomap_updater.cpp: In member function ‘virtual bool occupancy_map_monitor::DepthImageOctomapUpdater::initialize(const SharedPtr&)’:
/home/wislab/ros_robot/src/moveit2/moveit_ros/perception/depth_image_octomap_updater/src/depth_image_octomap_updater.cpp:119:59: error: no matching function for call to ‘occupancy_map_monitor::LazyFreeSpaceUpdater::LazyFreeSpaceUpdater(occupancy_map_monitor::OccMapTreePtr&)’
  119 |   free_space_updater_.reset(new LazyFreeSpaceUpdater(tree_));
      |                                                           ^
In file included from /home/wislab/ros_robot/src/moveit2/moveit_ros/perception/depth_image_octomap_updater/include/moveit/depth_image_octomap_updater/depth_image_octomap_updater.h:44,
                 from /home/wislab/ros_robot/src/moveit2/moveit_ros/perception/depth_image_octomap_updater/src/depth_image_octomap_updater.cpp:37:
/home/wislab/ros_robot/src/moveit2/moveit_ros/perception/lazy_free_space_updater/include/moveit/lazy_free_space_updater/lazy_free_space_updater.h:49:3: note: candidate: ‘occupancy_map_monitor::LazyFreeSpaceUpdater::LazyFreeSpaceUpdater(const OccMapTreePtr&, unsigned int)’
   49 |   LazyFreeSpaceUpdater(const collision_detection::OccMapTreePtr& tree, unsigned int max_batch_size = 10);
      |   ^~~~~~~~~~~~~~~~~~~~
/home/wislab/ros_robot/src/moveit2/moveit_ros/perception/lazy_free_space_updater/include/moveit/lazy_free_space_updater/lazy_free_space_updater.h:49:66: note:   no known conversion for argument 1 from ‘occupancy_map_monitor::OccMapTreePtr’ {aka ‘std::shared_ptr<occupancy_map_monitor::OccMapTree>’} to ‘const OccMapTreePtr&’ {aka ‘const std::shared_ptr<collision_detection::OccMapTree>&’}
   49 |   LazyFreeSpaceUpdater(const collision_detection::OccMapTreePtr& tree, unsigned int max_batch_size = 10);
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
make[2]: *** [depth_image_octomap_updater/CMakeFiles/moveit_depth_image_octomap_updater_core.dir/build.make:63: depth_image_octomap_updater/CMakeFiles/moveit_depth_image_octomap_updater_core.dir/src/depth_image_octomap_updater.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:425: depth_image_octomap_updater/CMakeFiles/moveit_depth_image_octomap_updater_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
Failed   <<< moveit_ros_perception [31.5s, exited with code 2]
Aborted  <<< moveit_ros_robot_interaction [36.4s]
Aborted  <<< moveit_kinematics [37.3s]
Aborted  <<< moveit_planners_ompl [42.9s]
Aborted  <<< moveit_ros_warehouse [43.6s]

YongLD, 2021-09-29:
I have the moveit2 binaries (Galactic) installed before, did it conflict?  And then I have the moveit2 source build by the steps of UniversalRobots/Universal_Robots_ROS2_Driver. 

Before this bug, I had fix the bug as follow:
`planning_scene_monitor.cpp:399:28:` Error: ‘collision_detection::OccMapTree’ has not been declared`.
> /home/wislab/ros_robot/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:399:28: error: ‘collision_detection::OccMapTree’ has not been declared
  399 |       collision_detection::OccMapTree::ReadLock lock;
      |                            ^~~~~~~~~~
/home/wislab/ros_robot/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:401:9: error: ‘lock’ was not declared in this scope
  401 |         lock = octomap_monitor_->getOcTreePtr()->reading();

### Solution: 
Add the header file `occupancy_map.h`  in planning_scene_monitor.cpp

`#include <moveit/collision_detection/occupancy_map.h>`

Did it matter?

YongLD, 2021-09-29:
@JafarAbdi

JafarAbdi, 2021-09-30:
Yes, having the binaries and the source may cause problems due a bug in amet_cmake, I would recommend removing moveit binaries ` sudo apt remove --purge ros-galactic-moveit*` and try building everything from source

YongLD, 2021-10-04:
@JafarAbdi Yes, it work! Thank you for the solution.

closed 2021-10-04 by YongLD