moveit/moveit2 #717: Replace the idx variable in TOTG plugin with the JointGroup API

AndyZe, 2021-10-02:
There was a [comment here](https://github.com/ros-planning/moveit2/pull/712#discussion_r718714860) on the Ruckig trajectory plugin PR which also applies to the "Time optmal trajectory generation" plugin:

> 
The code would be much more readable and consistent with MoveIt if you would use the joint group instead of passing num_dof and joint_idx everywhere.

Essentially, replace the usage of `idx` in the file moveit_core/trajectory_processing/src/time_optimal_trajectory_generation.cpp with usage of the Joint Group API.

invincialok, 2021-10-12:
i am a beginer how can i solve this issue . plz guide me

AndyZe, 2021-10-13:
Thanks @invincialok. I think what @henningkayser  meant is, you should go through the `moveit_core/trajectory_processing` folder and replace `num_dof` and `idx` arguments in places like this:

```
  static void getNextRuckigInput(const ruckig::OutputParameter<0>& ruckig_output,
                                 const moveit::core::RobotStatePtr& next_waypoint, size_t num_dof,
                                 const std::vector<int>& idx, ruckig::InputParameter<0>& ruckig_input);
```

With something like this:

```
  static void getNextRuckigInput(const ruckig::OutputParameter<0>& ruckig_output,
                                 const moveit::core::RobotStatePtr& next_waypoint,
                                 const moveit::core::JointModelGroup* joint_group,
                                 ruckig::InputParameter<0>& ruckig_input);
```

For example, you can replace `num_dof` with this:  `joint_group->getVariableCount();`

closed 2022-04-12 by AndyZe