moveit/moveit2 #726: Feature proposal: let the MoveItControllerManager manager more controllers

schornakj, 2021-10-07:
As currently implemented, the MoveItControllerManager plugin can only manage the activation and deactivation of controllers that have a MoveIt controller allocator plugin defined, due to [this bit of code](https://github.com/ros-planning/moveit2/blob/7292b9c20894dd65189d1ce72ee06981a520fd2f/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp#L153-L163).

However, it would also be useful to allow the MoveItControllerManager to also manage controllers that are not associated with MoveIt trajectory execution. For example, MoveIt Servo is often configured to use a streaming position or velocity controller, so switching between Servo control and joint trajectory execution requires managing these controllers separately from the execution of trajectories through the TrajectoryExecutionManager. The TrajectoryExecutionManager already has a lot of information about the relationships between different available controllers and their potentially-overlapping associations with joint interfaces, and it would be great to be able to use that more flexibly.

It's fairly straightforward to allow the MoveItControllerManager to manage more controllers -- the lines of code I linked above would need to be modified to be like this so the controller is added to `managed_controllers_` regardless of if a plugin is available for it:

```
std::string absname = getAbsName(controller.name);
auto controller_it = managed_controllers_.insert(std::make_pair(absname, controller)).first;  // with namespace
auto& claimed_interfaces = controller_it->second.claimed_interfaces;
std::transform(claimed_interfaces.cbegin(), claimed_interfaces.cend(), claimed_interfaces.begin(),
               [](const std::string& claimed_interface) {
                 return parseJointNameFromResource(claimed_interface);
               });

if (loader_.isClassAvailable(controller.type))
{
  allocate(absname, controller_it->second);
}
```

One way to use this expanded functionality would be to create a MoveGroup capability that activates a specified controller through the TrajectoryExecutionManager.anager.

Does anyone have thoughts about this?

tylerjw, 2021-10-07:
My only thought is we probably should have a better name.

https://blog.codinghorror.com/i-shall-call-it-somethingmanager/

I like the idea though.

tylerjw, 2021-11-02:
Hey team! Please [add your planning poker estimate](https://app.zenhub.com/workspaces/MoveIt-61675936b391800012280f6d/issues/ros-planning/moveit2/726?planning-poker) with ZenHub @abake48 @Abishalini @AndyZe @DLu @henningkayser @JafarAbdi @vatanaksoytezer

schornakj, 2021-11-05:
> Yes to all of that! For MTC my plan is to specify specific controllers via [external properties](ros-planning/moveit_task_constructor#194), but that support is not ready for merge yet.

Can you provide some detail about how you picture specifying and using specific controllers through an MTC task? Currently I use the `ExecuteTaskSolutionCapability` in my project, but since this ties me to the MoveGroup action interface I'm limited by the absence of a message field to describe my desired controller configuration.

> Also, when working with compliant controllers, it's far from clear when trajectory execution should finish or abort during execution.

One thing I was thinking about today was switching between compliant and non-compliant controllers that both control the same arm while executing a multi-segment trajectory. It's likely that at the end of a motion the compliant controller would leave the robot at a different joint state than originally planned, which would cause problems for the non-compliant controller since the start start of its trajectory would now be different than what was planned. This might necessitate a replan, and I don't have a sense for if there is existing support for this kind of runtime replanning in MoveIt. [Hybrid planning](https://github.com/ros-planning/moveit2/pull/763) could help solve this, though.

sjahr, 2021-11-10:
@schornakj I think Hybrid Planning could help with this problem because you could use the local planner to decouple global planning from the controller

closed 2021-12-07 by tylerjw

JaisonJose241, 2023-08-20:
Any updates? Has it been added onto MTC?