moveit/moveit2 #770: Error with Servo launching

AndyZe, 2021-11-01:
Commit 7e3fdcffb seems to have caused an issue with Servo launching. The robot joint state is not initialized and the robot is upright rather than having a 90* bend at the elbow. I tested with this command on Rolling branch:

`ros2 launch moveit_servo servo_example.launch.py`

No obvious errors in the terminal.

[image]

AndyZe, 2021-11-02:
PR #771 partially fixed this but wait, there's more:

`Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?`

The joint angles look good but the robot still does not move.

tylerjw, 2021-11-02:
Locally testing I got it working by doing this:

Start servo with the above launch command:

```
ros2 launch moveit_servo servo_example.launch.py
```

Call the start service:

```
ros2 service call /servo_node/start_servo std_srvs/srv/Trigger
```

Change the topics in servo_keyboard_input.cpp from moveit2_tutorials from `servo_server` to `servo_node` namespace as the namespace of the topics changed.  Then launch it with:

```
ros2 run moveit2_tutorials servo_keyboard_input
```

I'll submit a PR to fix that in the tutorial.  Is there some other way you are running it where it is not moving?  The error about providePlanningService seems unrelated to me but I will also submit a PR to fix that here too (looking back that's been there for a while).

tylerjw, 2021-11-02:
Here is the update for the tutorials: https://github.com/ros-planning/moveit2_tutorials/pull/176

tylerjw, 2021-11-02:
Note that you still need to call the start service with something.  I hope to fix the racing condition that prevented me from removing the start service altogether.

AndyZe, 2021-11-02:
Yep, I forgot that service call is necessary in ROS2. [This PR](https://github.com/ros-planning/moveit2_tutorials/pull/109) updated the documentation.

Yay, let's move on.

closed 2021-11-02 by AndyZe