moveit/moveit2 #772: Servo not timing out properly

brennand, 2021-11-01:
### Description

When there is a miss match in the control_msgs - JointJog time stamp, the the joint doesn't time out.

I think the time out is based on the time the msg is received.  I found this problem when I hadn't set the command line publisher to use sim time.

I think as this is a "safety feature"  it should use the time.now() for then the msg was received.

Also of note.  The duration field hasn't got any documentation and doesn't seem todo anything.  Please either updated the Documents for remove this field if not used.

### Your environment
* ROS Distro: ROS2 Rolling
* OS Version: 20.04
* Binary build?

### Steps to reproduce

Start Moviet / servo with sim time then send a command line msg:

ros2 topic pub /servo_server/delta_joint_cmds control_msgs/JointJog "{joint_names: ['elbow_joint'], displacements: [], velocities: [-1.0], duration: 0.0}"

### Expected behaviour
The joint moves for 0.1 sec then stops

### Actual behaviour
The joint continues moving for ever.

tylerjw, 2021-11-02:
Hey team! Please [add your planning poker estimate](https://app.zenhub.com/workspaces/MoveIt-61675936b391800012280f6d/issues/ros-planning/moveit2/772?planning-poker) with ZenHub @henningkayser @AndyZe @vatanaksoytezer @JafarAbdi @DLu @abake48 @Abishalini

tylerjw, 2021-11-24:
@brennand would you mind testing the PR I linked to see if it resolves this issue?  I would also appreciate a review of it if you have time.

https://github.com/ros-planning/moveit2/pull/825

henningkayser, 2023-01-26:
Has this been addressed in the meantime? I know that there have been several fixes to Servo runtime last year. @AndyZe any thoughts?

AndyZe, 2023-01-31:
I think timeouts are handled well if the incoming messages have a stamp. But @brennand has a good point, what happens if the message does not include a stamp? That isn't defined.

Sorry it took me so long to look at this issue.

AndyZe, 2023-01-31:
I just tested with [the tutorial](https://moveit.picknik.ai/main/doc/examples/realtime_servo/realtime_servo_tutorial.html) and both stamped and unstamped commands seem to behave in safe ways.

If the message is stamped, the timeout works as expected:

```
  if (msg->header.stamp != rclcpp::Time(0.))
    latest_joint_command_stamp_ = msg->header.stamp;
```

If the message isn't stamped, then `latest_joint_command_stamp_` remains where it was initialized (rclcpp::Time(0.)). rclcpp::Time can't be negative and it times out immediately -- meaning the robot never moves. So I'd say it's safe to close this.

closed 2023-01-31 by AndyZe