moveit/moveit2 #775: Unable to get MoveGroupInterface instance to work when used inside of a rclcpp::Node class.

larsbes, 2021-11-02:
### Description

In a larger framework of ROS2 nodes aimed at the operation of a robotic manipulator, I am trying to create a (set of) node(s) that is able to take instructions, pass them to MoveIt/move_group, and then send the obtained plan to the robot. To do so, I am using `moveit::planning_interface::MoveGroupInterface` to create an instance of `move_group`, for which a spinning node is required as argument. However, when using this interface in a class that inherits `rclcpp::Node`, and passing the spinning Node-class to `move_group`, the instance does not behave as if it is assigned a spinning node. The most prevalent error is "Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.". 

When using the RViz GUI, I am able to succesfully plan using the "plan" button, so I am assuming there are no problems with ros_controllers or the urdf/srdf. 

Below, I've attached the two nodes I have created. The first node is the main node that is the client of the service and sends planning instructions to the server. The second node is the server, and contains the move_group functionality. The second node is where the problem occurs. Below the code, I attached the output from MoveIt. 

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Binary
* Release version 6.1

### Steps to reproduce

Main node, client that requests a plan to be generated by move_group.

```
#define BOOST_BIND_NO_PLACEHOLDERS

#include "rclcpp/rclcpp.hpp"
#include "example_interfaces/srv/add_two_ints.hpp"
#include "std_msgs/msg/string.hpp"
#include "trajectory_msgs/msg/joint_trajectory.hpp"
#include "interfaces/msg/plan_instructions.hpp"
#include "interfaces/srv/plan_request.hpp"

#include <chrono>
#include <cstdlib>
#include <memory>
#include <string>
#include <queue>
#include <functional>

using namespace std::chrono_literals;
using std::placeholders::_1;

class MotionPlannerNode : public rclcpp::Node
{
public:
  MotionPlannerNode()
  : Node("node_motion_planner")
  {
    // publisher sends plan to CC, subscriber receives instr from CC, 
    // client communicates with node_moveit for plan generation
    publisher_ = this->create_publisher<trajectory_msgs::msg::JointTrajectory>("planner/new_plan", 10);
    subscription_ = this->create_subscription<interfaces::msg::PlanInstructions>(
      "planning_instructions", 10, std::bind(&MotionPlannerNode::queue_instructions, this, _1));
    client_ = service_node->create_client<interfaces::srv::PlanRequest>("move_group_functionality");
  }

  // sends plan to CC
  void send_plan(trajectory_msgs::msg::JointTrajectory plan_msg){ 
    publisher_->publish(plan_msg);
    RCLCPP_INFO(this->get_logger(), "I sent a plan", i);
    }

  void run()
  {
    if(!to_plan_points.empty()) {
      // pop front of the queue to obtain a set of instructions
      interfaces::msg::PlanInstructions instr = to_plan_points.front();

      // create request and fill it with the instructions
      auto request = std::make_shared<interfaces::srv::PlanRequest::Request>();
      request->instr = instr;

      // request a plan from node_moveit
      auto result = client_->async_send_request(request);
[... 320 lines cut ...]
[rviz2-1] [INFO] [1635848842.331128891] [moveit_ros_visualization.motion_planning_frame_planning]: planner id: arm[SPARStwo]
[rviz2-1] [INFO] [1635848842.331132797] [moveit_ros_visualization.motion_planning_frame_planning]: planner id: arm[STRIDE]
[rviz2-1] [INFO] [1635848842.331136616] [moveit_ros_visualization.motion_planning_frame_planning]: planner id: arm[TRRT]
[rviz2-1] [INFO] [1635848842.367304335] [interactive_marker_display_94337578632656]: Service response received for initialization
[node_motion_planner-7] [INFO] [1635848853.113461447] [node_motion_planner]: I got instructions
[node_motion_planner-7] not empty!
[node_motion_planner-7] before sending request
[node_motion_planner-7] after sending request
[node_moveit-6] Parsing robot urdf xml string.
[node_moveit-6] [INFO] [1635848853.123545156] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00169662 seconds
[node_moveit-6] [INFO] [1635848853.123596780] [moveit_robot_model.robot_model]: Loading robot model 'scara'...
[node_moveit-6] Link z_carriage had 1 children
[node_moveit-6] Link inner_link had 1 children
[node_moveit-6] Link outer_link had 1 children
[node_moveit-6] Link tcp had 0 children
[node_moveit-6] [INFO] [1635848853.471971068] [move_group_interface]: Ready to take commands for planning group arm.
[node_moveit-6] which one is stopping the func
[node_moveit-6] setting start and target poses
[node_moveit-6] pose target set
[node_moveit-6] [INFO] [1635848853.478285622] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[node_moveit-6] [INFO] [1635848854.478410438] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[node_moveit-6] Check clock synchronization if your are running ROS across multiple machines!
[node_moveit-6] [ERROR] [1635848854.478441893] [move_group_interface]: Failed to fetch current robot state
[ERROR] [node_moveit-6]: process has died [pid 24033, exit code -11, cmd '/home/larsb/vdl_ws/install/robot_motion_planner/lib/robot_motion_planner/node_moveit --ros-args --params-file /tmp/launch_params_jp4gmwcc --params-file /tmp/launch_params_5t4k8brz --params-file /tmp/launch_params_andorkvd --params-file /tmp/launch_params_z1lep7b1 --params-file /tmp/launch_params__lj2jmap --params-file /tmp/launch_params_0p_6x6uo --params-file /tmp/launch_params_tq3w0j0t'].
```

JafarAbdi, 2021-11-03:
Do you mind test changing the executor type to multi-thread .? also your program is segfaulting

```
[ERROR] [node_moveit-6]: process has died [pid 24033, exit code -11, cmd '/home/larsb/vdl_ws/install/robot_motion_planner/lib/robot_motion_planner/node_moveit --ros-args --params-file /tmp/launch_params_jp4gmwcc --params-file /tmp/launch_params_5t4k8brz --params-file /tmp/launch_params_andorkvd --params-file /tmp/launch_params_z1lep7b1 --params-file /tmp/launch_params__lj2jmap --params-file /tmp/launch_params_0p_6x6uo --params-file /tmp/launch_params_tq3w0j0t'].
```

and the problem lies here, you will have a dangling reference to `executor` as soon as you exit the `generate_plan` callback 

```cpp
rclcpp::executors::SingleThreadedExecutor executor;
executor.add_node(move_group_node);
std::thread([&executor]() { executor.spin(); }).detach();
```

JafarAbdi, 2021-12-13:
Closing due to inactivity, please feel free to reopen if you still have this problem

closed 2021-12-13 by JafarAbdi

RoboHack04, 2022-05-24:
@JafarAbdi  I am facing a similar issue, so what is the best way to spin the node and receive proper states from the current state monitor. 

p.s: I have tried to spin the node with rclcpp::spin(), and everything seems to work as expected but will run into warn/err from moveit_ros.current_state_monitor of "No state update received within 100ms of system clock".

zeidk, 2023-04-17:
@RoboHack04 Did you solve this issue?

fzoric8, 2024-05-09:
Did anyone solve this issue? 

I'm also not able to get robot state. 
You can see more [here](https://robotics.stackexchange.com/questions/111134/moveit2-get-robot-state-fails-because-of-the-different-times).

zhangpeng66, 2025-04-02:
ros2 foxy release, I'm also not able to get robot state by use move goup intetface