moveit/moveit2 #780: std::bad_alloc in move_group when launching UR Driver

JoNeckel, 2021-11-04:
### Description

Overview of your issue here.
I try to start MoveIt with the Universal_Robots_ROS2_Driver, but there is an exception thrown in the move_group node, I tried different Versions of MoveIt and also the Driver and other dependencies. But I am not yet an expert with this and couldn't find a solution. 

### Your environment
* ROS Distro: foxy
* OS Version: e.g. Ubuntu 20.04
* Source and Binary build (both tested)
* If binary, which release version? -> what you currently get here: https://moveit.ros.org/install-moveit2/binary/
* If source, which branch? -> foxy branch

### Steps to reproduce
follow the instructions of "Getting Started" and "Using MoveIt" of this Driver for Universal Robots: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/foxy

then launching the controllers:
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false`

and in another Terminal launch MoveIt
`ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur5e robot_ip:="xxx.xxx" use_fake_hardware:=true launch_rviz:=true`

### Expected behaviour
Rviz should Start with the Universal Robot Arm and you can plan and execute Motions there (without Hardware).
This has been working on Oct 12 but since i rebuilt everything a few days ago, it doesn't work anymore.. even with using an old Version of the Driver, so the Problem has to be somewhere else..

### Actual behaviour
Rviz starts, but without a planning library loaded, in the Terminal you get this output:
```
ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur5e robot_ip:="xxx.xxx" use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/jo/.ros/log/2021-10-28-18-57-00-505520-myubuntu-38988
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [38995]
[INFO] [mongo_wrapper_ros.py-2]: process started with pid [38997]
[INFO] [rviz2-3]: process started with pid [38999]
[INFO] [static_transform_publisher-4]: process started with pid [39001]
[static_transform_publisher-4] [INFO] [1635440221.371882197] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'base_link'
[move_group-1] [WARN] [1635440221.799056047] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1635440221.799131393] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Parsing robot urdf xml string.
[move_group-1] [INFO] [1635440221.806328362] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00255315 seconds
[move_group-1] [INFO] [1635440221.806395974] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[move_group-1] terminate called after throwing an instance of 'std::bad_alloc'
[move_group-1]   what():  std::bad_alloc
[mongo_wrapper_ros.py-2] [INFO] [1635440221.848078191] [mongodb]: /tmp/db does not exist; creating it.
[mongo_wrapper_ros.py-2] [INFO] [1635440221.848985167] [mongodb]: Starting mongodb with db location /tmp/db listening on localhost:33829
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] MongoDB starting : pid=39049 port=33829 dbpath=/tmp/db 64-bit host=myubuntu
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] db version v3.6.8
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] allocator: tcmalloc
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] modules: none
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] build environment:
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten]     distarch: x86_64
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten]     target_arch: x86_64
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I CONTROL  [initandlisten] options: { net: { port: 33829 }, storage: { dbPath: "/tmp/db" } }
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I STORAGE  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I STORAGE  [initandlisten] ** WARNING: Using the XFS filesystem is strongly recommended with the WiredTiger storage engine
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I STORAGE  [initandlisten] **          See http://dochub.mongodb.org/core/prodnotes-filesystem
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:01.895+0200 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=7438M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
[ERROR] [move_group-1]: process has died [pid 38995, exit code -6, cmd '/home/jo/workspace2/ros_ur_driver/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_cp1516ff --params-file /tmp/launch_params_5n71vdhp --params-file /tmp/launch_params_n22nlzd8 --params-file /tmp/launch_params_a9tldbh3 --params-file /tmp/launch_params_eu70qffg --params-file /tmp/launch_params_a33px69o --params-file /tmp/launch_params_sl6mfogw'].
[rviz2-3] [INFO] [1635440222.469861788] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1635440222.469966764] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1635440222.510429606] [rviz2]: Stereo is NOT SUPPORTED
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.542+0200 I STORAGE  [initandlisten] WiredTiger message [1635440222:542889][39049:0x7f287f99ff40], txn-recover: Set global recovery timestamp: 0
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] ** WARNING: This server is bound to localhost.
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] **          Remote systems will be unable to connect to this server. 
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] **          Start the server with --bind_ip <address> to specify which IP 
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] **          addresses it should serve responses from, or with --bind_ip_all to
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] **          bind to all interfaces. If this behavior is desired, start the
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] **          server with --bind_ip 127.0.0.1 to disable this warning.
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.606+0200 I STORAGE  [initandlisten] createCollection: admin.system.version with provided UUID: 4552777f-823e-4136-b2bb-1aacd8d2735e
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.639+0200 I COMMAND  [initandlisten] setting featureCompatibilityVersion to 3.6
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.642+0200 I STORAGE  [initandlisten] createCollection: local.startup_log with generated UUID: 3c72c821-80da-4b20-9bba-e83e4e6e8e19
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.665+0200 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '/tmp/db/diagnostic.data'
[mongo_wrapper_ros.py-2] 2021-10-28T18:57:02.666+0200 I NETWORK  [initandlisten] waiting for connections on port 33829
[... cut ...]
[rviz2-3] [ERROR] [1635440225.780069617] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] [INFO] [1635440225.891697207] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00228706 seconds
[rviz2-3] [INFO] [1635440225.892683553] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[rviz2-3] [ERROR] [1635440225.912147061] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': std::bad_alloc
```

JafarAbdi, 2021-11-04:
This got fixed with https://github.com/ros-planning/moveit2/pull/525, I'll backport it to foxy

JafarAbdi, 2021-11-04:
@JoNeckel Could you please test https://github.com/ros-planning/moveit2/pull/782 ? thanks!

JoNeckel, 2021-11-04:
Uno momento, I will test it, thanks for your fast reply!

JoNeckel, 2021-11-04:
@JafarAbdi unfortunately it didn't solve the problem. 
I just cloned the foxy branch, did I miss something?

JoNeckel, 2021-11-04:
thanks, I am building it again

JoNeckel, 2021-11-04:
Now i got an error while building:
```
~/workspace/ros_ur_driver$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Starting >>> moveit_common
Starting >>> warehouse_ros
Starting >>> moveit_resources_panda_description
Finished <<< moveit_common [1.67s]
Starting >>> moveit_resources_fanuc_description
Finished <<< ros2_control_test_assets [1.70s]
Starting >>> hardware_interface
Finished <<< moveit_resources_panda_description [1.71s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [3.22s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_moveit_config [3.22s]
Starting >>> moveit_resources_pr2_description
[Processing: controller_manager_msgs, geometric_shapes, hardware_interface, moveit_msgs, moveit_resources_fanuc_moveit_config, moveit_resources_pr2_description, srdfdom, warehouse_ros]
Finished <<< moveit_resources_fanuc_moveit_config [40.0s]
Finished <<< moveit_resources_pr2_description [40.0s]
Starting >>> moveit_resources
Finished <<< srdfdom [45.1s]
Finished <<< warehouse_ros [45.8s]
Starting >>> warehouse_ros_mongo
Finished <<< moveit_resources [1.68s]
Finished <<< controller_manager_msgs [1min 1s]
Finished <<< warehouse_ros_mongo [38.9s]
Finished <<< hardware_interface [1min 29s]
Starting >>> controller_interface
Starting >>> transmission_interface
[Processing: controller_interface, geometric_shapes, moveit_msgs, transmission_interface]
Finished <<< transmission_interface [37.5s]
Finished <<< controller_interface [38.8s]
Starting >>> controller_manager
[Processing: controller_manager, geometric_shapes, moveit_msgs]
Finished <<< geometric_shapes [2min 43s]
[Processing: controller_manager, moveit_msgs]
Finished <<< controller_manager [1min 7s]
Starting >>> forward_command_controller
Starting >>> joint_state_broadcaster
Starting >>> joint_trajectory_controller
Starting >>> diff_drive_controller
Starting >>> force_torque_sensor_broadcaster
Starting >>> gripper_controllers
Starting >>> imu_sensor_broadcaster
[Processing: diff_drive_controller, force_torque_sensor_broadcaster, forward_command_controller, gripper_controllers, imu_sensor_broadcaster, joint_state_broadcaster, joint_trajectory_controller, moveit_msgs]
[Processing: diff_drive_controller, force_torque_sensor_broadcaster, forward_command_controller, gripper_controllers, imu_sensor_broadcaster, joint_state_broadcaster, joint_trajectory_controller, moveit_msgs]
Finished <<< gripper_controllers [1min 8s]
Starting >>> ros2controlcli
Finished <<< forward_command_controller [1min 8s]
Starting >>> effort_controllers
Finished <<< imu_sensor_broadcaster [1min 8s]
Starting >>> position_controllers
Finished <<< ros2controlcli [3.53s]
Starting >>> velocity_controllers
Finished <<< force_torque_sensor_broadcaster [1min 15s]
Starting >>> ros2_control
Finished <<< joint_state_broadcaster [1min 17s]
Starting >>> joint_state_controller
Finished <<< moveit_msgs [4min 35s]
Starting >>> moveit_core
Finished <<< ros2_control [8.44s]
[Processing: diff_drive_controller, effort_controllers, joint_state_controller, joint_trajectory_controller, moveit_core, position_controllers, velocity_controllers]
Finished <<< diff_drive_controller [1min 59s]
Finished <<< joint_state_controller [41.8s]
Finished <<< joint_trajectory_controller [1min 59s]
[... 59 lines cut ...]
  121 |   std::scoped_lock slock(s.lock_);
      |        ^~~~~~~~~~~
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp:121:8: note: ‘std::scoped_lock’ is only available from C++17 onwards
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp: In function ‘moveit::core::RobotModelConstPtr moveit::planning_interface::getSharedRobotModel(const SharedPtr&, const string&)’:
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp:139:8: error: ‘scoped_lock’ is not a member of ‘std’
  139 |   std::scoped_lock slock(s.lock_);
      |        ^~~~~~~~~~~
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp:139:8: note: ‘std::scoped_lock’ is only available from C++17 onwards
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp: In function ‘planning_scene_monitor::CurrentStateMonitorPtr moveit::planning_interface::getSharedStateMonitor(const SharedPtr&, const RobotModelConstPtr&, const std::shared_ptr<tf2_ros::Buffer>&)’:
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp:157:8: error: ‘scoped_lock’ is not a member of ‘std’
  157 |   std::scoped_lock slock(s.lock_);
      |        ^~~~~~~~~~~
/home/jo/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp:157:8: note: ‘std::scoped_lock’ is only available from C++17 onwards
[... cut ...] .cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:192: common_planning_interface_objects/CMakeFiles/moveit_common_planning_interface_objects.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_ros_planning_interface [19.7s, exited with code 2]

Summary: 45 packages finished [17min 9s]
  1 package failed: moveit_ros_planning_interface
  1 package had stderr output: moveit_ros_planning_interface
  8 packages not processed
```

JafarAbdi, 2021-11-04:
I fixed the build error, do you mind fetching the latest update .?

JoNeckel, 2021-11-04:
ah okay nice, again with `gh pr checkout 782`?

JafarAbdi, 2021-11-04:
> ah okay nice, again with `gh pr checkout 782`?

Yes, I think this should do it for you

JoNeckel, 2021-11-04:
I still have the same error.. :(

JoNeckel, 2021-11-05:
@JafarAbdi I tried it again and now it works! Seems like yesterday i didn't removed everything properly or sth.
Thank you very much for fixing! :)

makeadl, 2021-11-05:
I still have the same issue after using:

git remote add foxy_fix https://github.com/JafarAbdi/moveit2.git && git fetch foxy_fix pr-foxy_backports && git checkout pr-foxy_backports

my git status looks like:

workspace/ros_ur_driver/src/moveit2$ git status
On branch pr-foxy_backports
Your branch is up to date with 'foxy_fix/pr-foxy_backports'.

nothing to commit, working tree clean

should I remove all these build log and install folder and build it again ?

Note: I did the git remote command inside the moveit2 folder. is that right ?

JafarAbdi, 2021-11-08:
> I still have the same issue after using:
>
> git remote add foxy_fix https://github.com/JafarAbdi/moveit2.git && git fetch foxy_fix pr-foxy_backports && git checkout pr-foxy_backports
>
> my git status [...]

Could you please try cleaning the workspace and test again .?

JafarAbdi, 2021-11-08:
Fixed with https://github.com/ros-planning/moveit2/pull/782

closed 2021-11-08 by JafarAbdi