moveit/moveit2 #783: CI job 'rolling-ci + ccov' broken for PRs (cached only)

henningkayser, 2021-11-04:
The CI job "rolling-ci + ccov" currently fails for pull requests with the following build error:

```
   --- stderr: moveit_ros_warehouse
  /usr/bin/ld: /home/runner/work/moveit2/moveit2/.work/upstream_ws/install/warehouse_ros/lib/libwarehouse_ros.so: undefined reference to `rclcpp::SerializedMessage::SerializedMessage(unsigned long, rcutils_allocator_t const&)'
  collect2: error: ld returned 1 exit status
  make[2]: *** [warehouse/CMakeFiles/moveit_warehouse_save_as_text.dir/build.make:318: warehouse/moveit_warehouse_save_as_text] Error 1
  make[1]: *** [CMakeFiles/Makefile2:162: warehouse/CMakeFiles/moveit_warehouse_save_as_text.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
  /usr/bin/ld: /home/runner/work/moveit2/moveit2/.work/upstream_ws/install/warehouse_ros/lib/libwarehouse_ros.so: undefined reference to `rclcpp::SerializedMessage::SerializedMessage(unsigned long, rcutils_allocator_t const&)'
  collect2: error: ld returned 1 exit status
  make[2]: *** [warehouse/CMakeFiles/moveit_warehouse_broadcast.dir/build.make:318: warehouse/moveit_warehouse_broadcast] Error 1
  make[1]: *** [CMakeFiles/Makefile2:189: warehouse/CMakeFiles/moveit_warehouse_broadcast.dir/all] Error 2
  make: *** [Makefile:141: all] Error 2
  ---
  Failed   <<< moveit_ros_warehouse [56.5s, exited with code 2]
```

The main branch doesn' t have this issue, also [disabling the cache fixes the job](https://github.com/ros-planning/moveit2/pull/673/commits/4a7b8402d2d4c0d804a6eb53a77627846bd04752).

tylerjw, 2021-11-09:
@JafarAbdi is going to add the release date to the hash of the upstream workspace to fix this.

tylerjw, 2021-12-13:
@JafarAbdi would you mind tackling this?  You created something for checking the date of the upstream (installed from debain) workspace already, correct?

closed 2022-01-03 by tylerjw