moveit/moveit2 #786: launch run_move_group from demo_nodes failing

gautz, 2021-11-09:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Rolling]
* OS Version: e.g. Ubuntu 20.04

### Steps to reproduce
`ros2 launch run_move_group run_move_group.launch.py`
or 
`ros2 launch run_ompl_constrained_planning run_move_group.launch.py`

### Expected behaviour
Panda robot loading

### Actual behaviour
no Panda robot visualization

### Backtrace or Console output

```
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-1]          at line 253 in /opt/ros/rolling/include/class_loader/class_loader_core.hpp
[rviz2-1] [ERROR] [1636467672.047736735] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1636467672.097239482] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00539682 seconds
[rviz2-1] [INFO] [1636467672.097384421] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-1] [INFO] [1636467672.387159113] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0013568 seconds
[rviz2-1] [INFO] [1636467672.387194531] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-1] [ERROR] [1636467672.394985128] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.panda_joint1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.panda_joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.
[rviz2-1] [INFO] [1636467672.396657662] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0016568 seconds
[rviz2-1] [INFO] [1636467672.396677590] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-1] [ERROR] [1636467672.401553525] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.panda_joint1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.panda_joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.
```
or 
```
[INFO] [launch]: All log files can be found below /home/gah/.ros/log/2021-11-09-15-30-19-476830-IPA-BN1338-Lin1-69214
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [69216]
[INFO] [static_transform_publisher-2]: process started with pid [69218]
[INFO] [robot_state_publisher-3]: process started with pid [69220]
[INFO] [move_group-4]: process started with pid [69222]
[INFO] [ros2_control_node-5]: process started with pid [69224]
[INFO] [mongo_wrapper_ros.py-6]: process started with pid [69226]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-7]: process started with pid [69228]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process started with pid [69233]
[static_transform_publisher-2] [WARN] [1636468220.090503588] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [INFO] [1636468220.101190507] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'panda_link0'
[robot_state_publisher-3] Link panda_link1 had 1 children
[robot_state_publisher-3] Link panda_link2 had 1 children
[robot_state_publisher-3] Link panda_link3 had 1 children
[robot_state_publisher-3] Link panda_link4 had 1 children
[robot_state_publisher-3] Link panda_link5 had 1 children
[robot_state_publisher-3] Link panda_link6 had 1 children
[robot_state_publisher-3] Link panda_link7 had 1 children
[robot_state_publisher-3] Link panda_link8 had 1 children
[robot_state_publisher-3] Link panda_hand had 2 children
[robot_state_publisher-3] Link panda_leftfinger had 0 children
[robot_state_publisher-3] Link panda_rightfinger had 0 children
[robot_state_publisher-3] [INFO] [1636468220.111941653] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-3] [INFO] [1636468220.112042444] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-3] [INFO] [1636468220.112062862] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-3] [INFO] [1636468220.112075557] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-3] [INFO] [1636468220.112086787] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-3] [INFO] [1636468220.112097829] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-3] [INFO] [1636468220.112108888] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-3] [INFO] [1636468220.112119709] [robot_state_publisher]: got segment panda_link5
[... 117 lines cut ...]
[move_group-4] *     - CartesianPathService
[move_group-4] *     - ExecuteTrajectoryAction
[move_group-4] *     - GetPlanningSceneService
[move_group-4] *     - KinematicsService
[move_group-4] *     - MoveAction
[move_group-4] *     - MotionPlanService
[move_group-4] *     - QueryPlannersService
[move_group-4] *     - StateValidationService
[move_group-4] ********************************************************
[move_group-4] 
[move_group-4] [INFO] [1636468235.826726908] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[move_group-4] [INFO] [1636468235.826736406] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context initialization complete
[move_group-4] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-4] Loading 'move_group/ClearOctomapService'...
[move_group-4] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-4] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-4] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-4] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-4] Loading 'move_group/MoveGroupMoveAction'...
[move_group-4] Loading 'move_group/MoveGroupPlanService'...
[move_group-4] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-4] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-4] 
[move_group-4] You can start planning now!
```

JafarAbdi, 2021-11-09:
The error is 
```
[rviz2-1] [ERROR] [1636468224.468571396] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.panda_joint1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.panda_joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.
```

Did you change anything in https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/joint_limits.yaml .? looks like you added double-quotes to the values which are causing them to be interpreted as strings

Also, are you using source build or binaries .?

gautz, 2021-11-09:
I  did not change anything, I am building moveit2 from source on rolling. Can you reproduce the issue?

JafarAbdi, 2021-11-09:
> I did not change anything, I am building moveit2 from source on rolling. Can you reproduce the issue?

I can't reproduce it, I'll try a new fresh build and report back to you

gautz, 2021-11-09:
interestingly `ros2 launch run_moveit_cpp run_moveit_cpp.launch.py` is working as expected.

vatanaksoytezer, 2021-11-09:
I couldn't reproduce this either with the latest docker or a fresh install. @gautz Do you mind making a clean build of the workspace and making sure you have the latest moveit and let us know how it goes?

gautz, 2021-11-09:
I am currently building the moveit `main` branch for rolling.
My previous workspace included `moveit_tutorials` `main` branch. Which `moveit_tutorials` branch is targeting rolling?

vatanaksoytezer, 2021-11-09:
You don't need  `moveit2_tutorials` for run_xxx tutorials. Having said that main branch of moveit2_tutorials is targeting rolling.

JafarAbdi, 2021-11-09:
I tried again with a clean build and it's working pretty well, do mind sharing the output of `apt policy ros-rolling-moveit-resources-panda-moveit-config` .?

gautz, 2021-11-09:
```
ros-rolling-moveit-resources-panda-moveit-config:
  Installed: 2.0.3-1focal.20211102.152149
  Candidate: 2.0.3-1focal.20211102.152149
  Version table:
 *** 2.0.3-1focal.20211102.152149 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```

henningkayser, 2021-12-16:
Please [add your planning poker estimate](https://app.zenhub.com/workspaces/MoveIt-61675936b391800012280f6d/issues/ros-planning/moveit2/786?planning-poker) with ZenHub @vatanaksoytezer

vatanaksoytezer, 2021-12-16:
These are now removed with #922. If you are still having failures on tutorials or anything else feel free to open a issue.

closed 2021-12-16 by vatanaksoytezer