moveit/moveit2 #796: std::bad_alloc in RobotModelLoader

henningkayser, 2021-11-12:
### Description

I frequently get a `std::bad_alloc` exception in the RobotModelLoader when launching MoveGroup, MoveItCpp and RViz (plugins).
Attached is a console output filtered for RViz and MoveGroup (this uses the prbt robot, but I see just the same with the panda and others).

Take note, that both RViz plugins (MotionPlanning, PlanningSceneDisplay) fail attempting to load the robot.
I identified the following line as the cause: https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/robot_model_loader/src/robot_model_loader.cpp#L125

When I remove `= joint_model->getVariableBoundsMsg()` and just initialize an empty vector, everything works fine. Curiously, I can query the size of the vector which appears to be a very very big number (need to look that up again), so here `std::bad_alloc` could actually be a case of insufficient memory for copy. To me it's just puzzling how this can be the case at all.

Side notes:
* Recompiled multiple times using clang and also gcc, with fresh workspace, cleared ccache

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 20.04
* Source build, main branch
* Compiler: gcc, clang

### Steps to reproduce
Build from source, run a demo: `ros2 launch moveit_resources_prbt_moveit_config demo.launch.py`

### Expected behaviour
It should initialize a robot model.

### Actual behaviour
I just get a lousy `std::bad_alloc`...

### Console output

```
[INFO] [move_group-4]: process started with pid [102470]
[move_group-4] [INFO] [1636732230.834732234] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00192166 seconds
[move_group-4] [INFO] [1636732230.834797003] [moveit_robot_model.robot_model]: Loading robot model 'prbt'...
[move_group-4] terminate called after throwing an instance of 'std::bad_alloc'
[move_group-4]   what():  std::bad_alloc

[INFO] [rviz2-1]: process started with pid [102464]
[rviz2-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ubuntu'
[rviz2-1] [INFO] [1636732230.921231477] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1636732230.921405356] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1636732230.936551123] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [ERROR] [1636732231.109095524] [moveit_background_processing.background_processing]: Begin executing 'loadRobotModel'
[rviz2-1] [INFO] [1636732231.114448190] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00167864 seconds
[rviz2-1] [INFO] [1636732231.114491860] [moveit_robot_model.robot_model]: Loading robot model 'prbt'...
[rviz2-1] [INFO] [1636732231.561246353] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00188578 seconds
[rviz2-1] [INFO] [1636732231.561291823] [moveit_robot_model.robot_model]: Loading robot model 'prbt'...
[rviz2-1] [ERROR] [1636732231.561654490] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': std::bad_alloc
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-1]          at line 261 in /opt/ros/rolling/include/class_loader/class_loader_core.hpp
[rviz2-1] [ERROR] [1636732240.527797554] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [ERROR] [1636732240.540288605] [moveit_background_processing.background_processing]: Begin executing 'loadRobotModel'
[rviz2-1] [INFO] [1636732240.575806909] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00238518 seconds
[rviz2-1] [INFO] [1636732240.575831959] [moveit_robot_model.robot_model]: Loading robot model 'prbt'...
[rviz2-1] [ERROR] [1636732240.576154646] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': std::bad_alloc
```

Briancbn, 2021-11-12:
I was also getting the same error in Foxy 2.2.1. I realized that **after** having installed the debian binary and then build moveit2 from source (tagged 2.2.1 and from scratch) this error will occur very consistently. My solution was

1. Uninstall binaries

   ```bash
   sudo apt install ros-foxy-moveit*
   ```
   Then remove build and install to rebuild, (just remove build is also not working).

2. Use only binary and don't source the install

I am not exactly sure if it is the same issue and I didn't do enough digging. Hope this also give some light on `bad_alloc` issue.

v4hn, 2021-11-12:
Sounds like the includes or libraries are wrongly preferred from the underlying workspace.
One way to find out where the violation occurs is by making all dev files from ros-foxy-moveit* unreadable(`-r`) and try to build.
The compiler will complain if it can't open an existing file that's preferred.
Just some thoughts.

henningkayser, 2021-11-13:
In my case (rolling) I don't have any MoveIt binaries installed, unfortunately.

JafarAbdi, 2021-11-22:
I had the same issue on my other PC and fixed it by running `sudo apt remove --purge ros-rolling-moveit*` then recompiling with a clean workspace, this's a bug in ament_cmake

JafarAbdi, 2021-12-13:
Closing see my previous comment

closed 2021-12-13 by JafarAbdi

Briancbn, 2021-12-13:
> I had the same issue on my other PC and fixed it by running `sudo apt remove --purge ros-rolling-moveit*` then recompiling with a clean workspace, this's a bug in ament_cmake

@JafarAbdi Is it possible to keep this open still to keep track of the issue, or link to the issue for the aforementioned bug? Although this is technically solved, this mentioned ament bug still causes some inconvenience in development, e.g. using MoveIt2 debian for one workspace/project while source build for the other.