moveit/moveit2 #802: Question move_group_interface xacro

JoNeckel, 2021-11-15:
Hello,
I want to use MoveIt and the move_group C++ interface with a Universal Robots UR10e.

I use ROS2 foxy.

And I have this Driver: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver working with the Rviz Plugin of MoveIt.

Now I wanted to change the panda robot to my UR to test this Tutorial: http://moveit2_tutorials.picknik.ai/doc/move_group_interface/move_group_interface_tutorial.html

But I can't get it to work... 

first I launched the controller, then the move_group so that I can plan and execute motions with the Rviz plugin.

then I wanted to launch the Tutorial 

`ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py`

which I built inside my UR workspace with following changes in the move_group_interface_tutorial.cpp:
in line 74: `  static const std::string PLANNING_GROUP = "ur_manipulator"; '
in line 93: ```
 moveit_visual_tools::MoveItVisualTools visual_tools(move_group_node, "base_link", "move_group_tutorial",
                                                      move_group.getRobotModel()); 

```
and in the launchfile, i made these changes:

```
import os
import yaml
from launch import LaunchDescription
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory

def load_file(package_name, file_path):
    package_path = get_package_share_directory(package_name)
    absolute_file_path = os.path.join(package_path, file_path)

    try:
        with open(absolute_file_path, "r") as file:
            return file.read()
    except EnvironmentError:  # parent of IOError, OSError *and* WindowsError where available
        return None

def load_yaml(package_name, file_path):
    package_path = get_package_share_directory(package_name)
    absolute_file_path = os.path.join(package_path, file_path)

    try:
        with open(absolute_file_path, "r") as file:
            return yaml.safe_load(file)
    except EnvironmentError:  # parent of IOError, OSError *and* WindowsError where available
        return None

def generate_launch_description():
    # planning_context
    robot_description_config = load_file("ur_description", "urdf/ur.urdf.xacro")

    
    robot_description = {"robot_description": robot_description_config}
    
    robot_description_semantic_config = load_file(
        "ur_moveit_config", "srdf/ur.srdf.xacro"
    )
    robot_description_semantic = {
        "robot_description_semantic": robot_description_semantic_config
    }

    kinematics_yaml = load_yaml(
        "ur_description", "config/ur10e/default_kinematics.yaml"
    )

    # MoveGroupInterface demo executable
    move_group_demo = Node(
        name="move_group_interface_tutorial",
        package="moveit2_tutorials",
        executable="move_group_interface_tutorial",
        output="screen",
        parameters=[robot_description , robot_description_semantic, kinematics_yaml],
    )

    return LaunchDescription([move_group_demo])
```

I get this output in the terminal:
```
jo@myubuntu:~/workspace/ros_ur_driver$ ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py
[INFO] [launch]: All log files can be found below /home/jo/.ros/log/2021-11-15-16-06-21-343425-myubuntu-192289
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group_interface_tutorial-1]: process started with pid [192291]
[move_group_interface_tutorial-1] [WARN] [1636988781.672854546] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[move_group_interface_tutorial-1] Parsing robot urdf xml string.
[move_group_interface_tutorial-1] Error:   No link elements found in urdf file
[move_group_interface_tutorial-1]          at line 179 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/model.cpp
[move_group_interface_tutorial-1] [INFO] [1636988781.677823879] [moveit_rdf_loader.rdf_loader]: Unable to parse URDF from parameter: 'robot_description'
[move_group_interface_tutorial-1] [FATAL] [1636988781.677896310] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
[move_group_interface_tutorial-1] terminate called after throwing an instance of 'std::runtime_error'
[move_group_interface_tutorial-1]   what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ERROR] [move_group_interface_tutorial-1]: process has died [pid 192291, exit code -6, cmd '/home/jo/workspace/ros_ur_driver/install/moveit2_tutorials/lib/moveit2_tutorials/move_group_interface_tutorial --ros-args -r __node:=move_group_interface_tutorial --params-file /tmp/launch_params_oiqua8yh --params-file /tmp/launch_params_x1406may --params-file /tmp/launch_params_4x0uu_6c'].
jo@myubuntu:~/workspace/ros_ur_driver$ 
```

Is it possible to load a xacro like this or do I have to change something if I use a .xacro instead of a .urdf?

JoNeckel, 2021-11-15:
The ur.urdf.xacro which i want to load looks like this:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="$(arg name)"/>
   <!-- robot name parameter -->
   <xacro:arg name="name" default="ur"/> 
   <!-- import main macro -->
   <xacro:include filename="$(find ur_description)/urdf/ur_macro.xacro"/>

   <!-- parameters -->
   <xacro:arg name="prefix" default="" />
   <xacro:arg name="joint_limit_params" default=""/>
   <xacro:arg name="kinematics_params" default=""/>
   <xacro:arg name="physical_params" default=""/>
   <xacro:arg name="visual_params" default=""/>
   <xacro:arg name="transmission_hw_interface" default=""/>
   <xacro:arg name="safety_limits" default="false"/>
   <xacro:arg name="safety_pos_margin" default="0.15"/>
   <xacro:arg name="safety_k_position" default="20"/>
   <!-- ros2_control related parameters -->
   <xacro:arg name="use_fake_hardware" default="false" />
   <xacro:arg name="fake_sensor_commands" default="false" />

   <!-- initial position -->
   <xacro:arg name="initial_positions_file" default="$(find ur_description)/config/initial_positions.yaml"/>
   <xacro:property name="initial_positions_file" default="$(arg initial_positions_file)"/>

   <!-- arm -->
   <xacro:ur_robot
     prefix="$(arg prefix)"
     joint_limits_parameters_file="$(arg joint_limit_params)"
     kinematics_parameters_file="$(arg kinematics_params)"
     physical_parameters_file="$(arg physical_params)"
     visual_parameters_file="$(arg visual_params)"
     transmission_hw_interface="$(arg transmission_hw_interface)"
     safety_limits="$(arg safety_limits)"
     safety_pos_margin="$(arg safety_pos_margin)"
     safety_k_position="$(arg safety_k_position)"
     use_fake_hardware="$(arg use_fake_hardware)"
     fake_sensor_commands="$(arg fake_sensor_commands)"
     initial_positions="${load_yaml(initial_positions_file)}" />
</robot>
```
I think maybe I need a parameter (like ur_type:=ur10e or something, at least thats what I need when launching the controllers and move_group) but I don't know how to set the parameter in the launchfile or anywhere else

tylerjw, 2021-11-15:
Hello, @JoNeckel I've messaged some people internally here who might be able to help you with this.  I know this launch file is for moveit_servo but the URDF parsing is probably the same.  You might be able to figure out what you need from this:

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_moveit_config/launch/test_moveit_servo_velocity_controller.launch.py

destogl, 2021-11-15:
@JoNeckel probably some parameters for the UR robots are missing there. Did you try running [ur_moveit.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/launch/ur_moveit.launch.py)? This should provide you with minimal example for UR.

If this is working, I rather propose to adapt the file from the UR-driver repository to avoid misconfiguring the UR-xacro. Can you please test this? @livanov93 do you have anything to add?

livanov93, 2021-11-15:
@JoNeckel You have to parse.urdf.xacro file with xacro as it is done [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/7c1e9b19c14e98b46662752f2ef4c1a23f1341d0/ur_bringup/launch/ur_control.launch.py#L73-L130).
.srdf.xacro file talso needs to be parsed with xacro. Take a look [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/7c1e9b19c14e98b46662752f2ef4c1a23f1341d0/ur_bringup/launch/ur_moveit.launch.py#L129-L146).
Generally, it is easy to misconfigure it, so the best way is to follow the launch files from the [official ROS2 driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main/ur_bringup/launch).

JoNeckel, 2021-11-15:
Thanks for your fast Answers! @tylerjw @destogl @livanov93
I think I am missing something really basic, I want to get a minimal example running where I can let the UR move to an cartesian point which I tell it via code instead of the rviz gui. 
I already made the ur_moveit.launch.py running, but what are the steps to be able to use the move_group_interface (C++)? 
when I try to run it directly, then it says this: (that's why i thought i need the launch file from the moveit tutorial in addition to the ur_bringup launchfiles)
```
jo@myubuntu:~/workspace/ros_ur_driver$ ros2 run moveit2_tutorials move_group_interface_tutorial 
[INFO] [1636997836.202430588] [moveit_rdf_loader.rdf_loader]: Robot model parameter not found! Did you remap 'robot_description'?

[FATAL] [1636997836.202536668] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
```

tylerjw, 2021-11-15:
Something is wrong with how you are passing in the robot description parameter.

Did you fix this error?

```
[move_group_interface_tutorial-1] Parsing robot urdf xml string.
[move_group_interface_tutorial-1] Error:   No link elements found in urdf file
[move_group_interface_tutorial-1]          at line 179 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/model.cpp
[move_group_interface_tutorial-1] [INFO] [1636988781.677823879] [moveit_rdf_loader.rdf_loader]: Unable to parse URDF from parameter: 'robot_description'
[move_group_interface_tutorial-1] [FATAL] [1636988781.677896310] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
```

It looks like before you were passing in something into that parameter that was failing to parse and maybe now you aren't passing anything into it?

JoNeckel, 2021-11-15:
No, I didn't fix it yet @tylerjw, I'm not sure if I need to fix that because i don't really know whats needed

tylerjw, 2021-11-15:
I would start here: `No link elements found in urdf file` and look for how your launch file differs from the ones linked in the official ROS2 driver for UR.

tylerjw, 2021-11-15:
The second error (`Unable to construct robot model. Please make sure all needed information is on the parameter server.
`) is just a consequence of the parsing of robot_description failing.

JoNeckel, 2021-11-15:
okay, I'll try that. I was thinking from the Answers, that the launch files from the Driver are enough. I'll keep you updated, thanks for your help!

JoNeckel, 2021-11-15:
Okay, it kind of works now, thank you! 
I took the ur_moveit.launch.py as a template and just added the move_group_interface - Executable and only started that node.
There are still some issues with the visualization, as you can see in this picture, but I hope that's something I'll figure out. 
[image]
Anyway, my original Issue is solved, thank you all!

closed 2021-11-15 by JoNeckel