moveit/moveit2 #803: Update REP2005 links

tylerjw, 2021-11-15:
### User Story

The links on REP 2005 (https://github.com/ros-infrastructure/rep/blob/master/rep-2005.rst) for MoveIt2 should follow the pattern of other projects and link to ros index.

### People

- Requester: @davetcoleman 
- Assignee: @stephanie-eng 

### Details:

MoveIt has sloppy links that should instead point to the individual packages on ROS index, like all the other entries in that document to. Could you make a PR please to point instead at index.ros.org instead of github?
Instead of:
    * `moveit2/moveit_jog_arm <https://github.com/ros-planning/moveit2/>`_
From this raw file
It should look more like:
  * `ament_cmake/ament_cmake <https://index.ros.org/p/ament_cmake/>`_
w.r.t. the URL

stephanie-eng, 2021-11-15:
I can update the links for most of the packages in the existing list, but I had some issues with the following:

- [moveit_jog_arm](https://index.ros.org/p/moveit_jog_arm/) seems to only be available on Eloquent
- [moveit_fake_controller_manager](https://index.ros.org/p/moveit_fake_controller_manager/) seems to be only available for ROS1
- [moveit_planners_trajopt](https://index.ros.org/p/moveit_planners_trajopt) seems to not exist on ROS Index, unless it is to be found under a different name I could be unaware of
  - The link above (what I think the link should be) returns 404

Also, from the ROS Index moveit repository page, there are packages that are not included in the list of links from the REP. Should any of these packages be added to the REP list?:

- [moveit_common](https://index.ros.org/p/moveit_common/)
- [run_move_group](https://index.ros.org/p/run_move_group/)
- [run_ompl_constrained_planning](https://index.ros.org/p/run_ompl_constrained_planning/)
- [pilz_industrial_motion_planner](https://index.ros.org/p/pilz_industrial_motion_planner/)
- [pilz_industrial_motion_planner_testutils](https://index.ros.org/p/pilz_industrial_motion_planner_testutils/)
- [sbpl_interface](https://index.ros.org/p/sbpl_interface/)
- [sbpl_interface_ros](https://index.ros.org/p/sbpl_interface_ros/)
- [moveit_servo](https://index.ros.org/p/moveit_servo/)

tylerjw, 2021-12-01:
@stephanie-eng it looks like the second one has conflicts.  Please update the PR to fix the conflicts.

stephanie-eng, 2021-12-02:
Should be resolved now.

closed 2021-12-16 by henningkayser