moveit/moveit2 #806: [Foxy] std::bad_alloc when running run_moveit_cpp demo

RoboticsYY, 2021-11-16:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Foxy]
* OS Version: e.g. Ubuntu 20.04
* Source build, `foxy` branch

### Steps to reproduce
Met `std::bad_alloc` with the latest Foxy source build. 

Roll back to commit `ee704d2f102cebfc6e0723d4c744f39fba3831c9`, resolve the issue.
### Console output
```shell
intel@intel-Tiger-Lake-Client-Platform:~/ws_moveit2$ ros2 launch run_moveit_cpp run_moveit_cpp.launch.py 
[INFO] [launch]: All log files can be found below /home/intel/.ros/log/2021-11-16-15-44-39-988147-intel-Tiger-Lake-Client-Platform-15016
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [15018]
[INFO] [robot_state_publisher-2]: process started with pid [15020]
[INFO] [rviz2-3]: process started with pid [15022]
[INFO] [run_moveit_cpp-4]: process started with pid [15024]
[INFO] [ros2_control_node-5]: process started with pid [15026]
[INFO] [ros2 run controller_manager spawner.py panda_arm_controller-6]: process started with pid [15028]
[INFO] [ros2 run controller_manager spawner.py panda_hand_controller-7]: process started with pid [15030]
[INFO] [ros2 run controller_manager spawner.py joint_state_broadcaster-8]: process started with pid [15032]
[static_transform_publisher-1] [INFO] [1637048680.402590229] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'panda_link0'
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link panda_link1 had 1 children
[robot_state_publisher-2] Link panda_link2 had 1 children
[robot_state_publisher-2] Link panda_link3 had 1 children
[robot_state_publisher-2] Link panda_link4 had 1 children
[robot_state_publisher-2] Link panda_link5 had 1 children
[robot_state_publisher-2] Link panda_link6 had 1 children
[robot_state_publisher-2] Link panda_link7 had 1 children
[robot_state_publisher-2] Link panda_link8 had 1 children
[robot_state_publisher-2] Link panda_hand had 2 children
[robot_state_publisher-2] Link panda_leftfinger had 0 children
[robot_state_publisher-2] Link panda_rightfinger had 0 children
[robot_state_publisher-2] [INFO] [1637048680.405160797] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-2] [INFO] [1637048680.405318445] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-2] [INFO] [1637048680.405329055] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-2] [INFO] [1637048680.405335724] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-2] [INFO] [1637048680.405341245] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-2] [INFO] [1637048680.405347161] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-2] [INFO] [1637048680.405352828] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-2] [INFO] [1637048680.405358414] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-2] [INFO] [1637048680.405364179] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-2] [INFO] [1637048680.405370216] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-2] [INFO] [1637048680.405375836] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-2] [INFO] [1637048680.405382482] [robot_state_publisher]: got segment panda_rightfinger
[run_moveit_cpp-4] [INFO] [1637048680.417252497] [moveit_cpp_demo]: Initialize node
[ros2_control_node-5] [INFO] [1637048680.440504025] [controller_manager]: update rate is 100 Hz
[ros2 run controller_manager spawner.py joint_state_broadcaster-8] [INFO] [1637048680.653899830] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[ros2 run controller_manager spawner.py panda_arm_controller-6] [INFO] [1637048680.698092710] [spawner_panda_arm_controller]: Waiting for /controller_manager services
[ros2 run controller_manager spawner.py panda_hand_controller-7] [INFO] [1637048680.698695835] [spawner_panda_hand_controller]: Waiting for /controller_manager services
[rviz2-3] [INFO] [1637048680.796940767] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1637048680.797114314] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1637048680.830576437] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-5] [INFO] [1637048680.865574957] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2 run controller_manager spawner.py joint_state_broadcaster-8] [INFO] [1637048680.875321459] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-5] [INFO] [1637048680.876363988] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2 run controller_manager spawner.py joint_state_broadcaster-8] [INFO] [1637048680.895660508] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[ros2_control_node-5] [INFO] [1637048680.905303581] [controller_manager]: Loading controller 'panda_arm_controller'
[ros2_control_node-5] [INFO] [1637048680.925210626] [controller_manager]: Loading controller 'panda_hand_controller'
[ros2 run controller_manager spawner.py panda_arm_controller-6] [INFO] [1637048680.926063414] [spawner_panda_arm_controller]: Loaded panda_arm_controller
[ros2_control_node-5] [INFO] [1637048680.935056524] [controller_manager]: Configuring controller 'panda_arm_controller'
[ros2_control_node-5] [INFO] [1637048680.935178857] [panda_arm_controller]: Command interfaces are [position] and and state interfaces are [position velocity].
[ros2_control_node-5] [INFO] [1637048680.935871273] [panda_arm_controller]: Controller state will be published at 50.00 Hz.
[ros2 run controller_manager spawner.py panda_hand_controller-7] [INFO] [1637048680.936405858] [spawner_panda_hand_controller]: Loaded panda_hand_controller
[... 7 lines cut ...]
[rviz2-3] [INFO] [1637048680.993394852] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00124008 seconds
[rviz2-3] [INFO] [1637048680.993830156] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[INFO] [ros2 run controller_manager spawner.py panda_arm_controller-6]: process has finished cleanly [pid 15028]
[INFO] [ros2 run controller_manager spawner.py panda_hand_controller-7]: process has finished cleanly [pid 15030]
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] [INFO] [1637048681.626273503] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.000734283 seconds
[rviz2-3] [INFO] [1637048681.626331321] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-3] [ERROR] [1637048681.632853083] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': std::bad_alloc
[run_moveit_cpp-4] [INFO] [1637048685.451817055] [moveit_cpp_demo]: Initialize MoveItCpp
[run_moveit_cpp-4] Parsing robot urdf xml string.
[run_moveit_cpp-4] [INFO] [1637048685.460250290] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00164579 seconds
[run_moveit_cpp-4] [INFO] [1637048685.460321735] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[run_moveit_cpp-4] terminate called after throwing an instance of 'std::bad_alloc'
[run_moveit_cpp-4]   what():  std::bad_alloc
[ERROR] [run_moveit_cpp-4]: process has died [pid 15024, exit code -6, cmd '/home/intel/ws_moveit2/install/run_moveit_cpp/lib/run_moveit_cpp/run_moveit_cpp --ros-args -r __node:=run_moveit_cpp --params-file /home/intel/ws_moveit2/install/run_moveit_cpp/share/run_moveit_cpp/config/moveit_cpp.yaml --params-file /tmp/launch_params_isgc0o1j --params-file /tmp/launch_params_paumwpb_ --params-file /tmp/launch_params_f1kw6cee --params-file /tmp/launch_params_ppoajui4 --params-file /tmp/launch_params_vuq3ml1x --params-file /tmp/launch_params_43nuuqkh'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ros2_control_node-5] [INFO] [1637048704.608642489] [rclcpp]: signal_handler(signal_value=2)
[rviz2-3] [INFO] [1637048704.608641818] [rclcpp]: signal_handler(signal_value=2)
[static_transform_publisher-1] [INFO] [1637048704.608644574] [rclcpp]: signal_handler(signal_value=2)
[robot_state_publisher-2] [INFO] [1637048704.608674233] [rclcpp]: signal_handler(signal_value=2)
[INFO] [static_transform_publisher-1]: process has finished cleanly [pid 15018]
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 15020]
[INFO] [ros2_control_node-5]: process has finished cleanly [pid 15026]
[INFO] [rviz2-3]: process has finished cleanly [pid 15022]
```

henningkayser, 2021-11-17:
Interesting! This could be related or even a duplicate of #796. Do you have the MoveIt 2 Foxy debians installed?

JafarAbdi, 2021-12-13:
This should no longer exist, please feel free to reopen if you still have this problem

closed 2021-12-13 by JafarAbdi