moveit/moveit2 #807: Moveit1 and Moveit2 Time parameterization

alexlt97, 2021-11-16:
Good afternoon,

We have encountered some difficulties trying to generate trajectories from Moveit2 for a robot that works in ROS1. We created the same trajectory in both versions of MoveIt but their results differ substantially. Here I attached the related issue:
https://github.com/Kinovarobotics/ros_kortex/issues/206#issuecomment-970178986

We suspect there is a difference in how Moveit2 handles time parameterization in creating trajectories.
It is possible to have an opinion about this?

AndyZe, 2021-11-16:
We could use some more information. In your `xxx_planning_pipeline.launch.xml` do you have a line like this:

      `default_planner_request_adapters/AddTimeParameterization`
      
 or is it `default_planner_request_adapters/AddTimeOptimalParameterization` ?
 
 I'd recommend AddTimeOptimalParameterization.
 
 It's also not clear to me what your graphs in the ros_kortex issue show. How did you get the data for those graphs? What part of the pipeline is it taken from?

AndyZe, 2021-11-16:
Do the graphs come from this topic:   `control_msgs/FollowJointTrajectoryActionGoalcontrol_msgs/FollowJointTrajectoryActionGoal`?

Thijsons, 2021-11-17:
Thank you for the response,

The issue was indeed a time parameterization one. The `AddTimeOptimalParameterization` had to be changed to `AddIterativeSplineParameterization` to work with the kinova driver (see the linked issue). This fixed the problem.

Thijsons, 2021-11-17:
@alexlt97 since the problem has been resolved the issue should probably be closed.

AndyZe, 2021-11-17:
I don't like to hear this -- `AddTimeOptimalParameterization` is our best option at the moment, so it should work. Will make a more detailed issue. Thanks for reporting this and providing a fair amount of data.

closed 2021-11-17 by AndyZe