moveit/moveit2 #814: Hybrid planning: eliminate timers in constructors

AndyZe, 2021-11-18:
The new hybrid planning feature uses timers in several constructors to wait for other things to initialize. For example in `global_planner/global_planner_component.cpp`.

Tyler made a comment here that might [help](https://github.com/ros-planning/moveit2/pull/488#discussion_r720408829).

sjahr, 2021-11-19:
I've already addressed this in #811 with 
- [a985b8c](https://github.com/ros-planning/moveit2/pull/811/commits/a985b8c3fc74f75551e74159e4849936bba55067) &#8594; removes timer in Global and Local Planner 
- [b027f62](https://github.com/ros-planning/moveit2/pull/811/commits/b027f625442b7f53091ff3b7af6f26bae7a3e9fd) and [5682263](https://github.com/ros-planning/moveit2/pull/811/commits/5682263a4fec2fe2f5efa404cec2f2f263e1c111) &#8594; removes timer Hybrid Planning Manager and refactors it (These two commits still cause a segfault which needs to be debugged)

henningkayser, 2021-11-19:
> I've already addressed this in #811 with
>
> * [a985b8c](https://github.com/ros-planning/moveit2/pull/811/commits/a985b8c3fc74f75551e74159e4849936bba55067) → removes timer in Global and Local Planner
> * [...]

I just wanted to say that. @tylerjw fixed it with https://github.com/ros-planning/moveit2/pull/763 and all the changes seem to have made it into #811.

AndyZe, 2021-11-19:
Tyler's branch crashes, though.

henningkayser, 2021-11-19:
That's probably not a good reason to drop his fixes. The issue you found seemed like a launch/configuration bug which should be pretty minor. Maybe @tylerjw can chime in and report on the latest state of his branch? It might save time if all of us would discuss this over a call or so.

AndyZe, 2021-11-19:
I'm not saying we drop the fixes. I'm saying we merge the 4000-line 200+ comment PR (which is currently functional) then we can apply the fixes in a follow-up.