moveit/moveit2 #818: Duplicate imports in Servo launch files

AndyZe, 2021-11-19:
Eliminate duplicate imports in Servo launch files, such as here in [servo_launch_test_common.py](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/test/launch/servo_launch_test_common.py):

```
import launch_ros
...
from launch_ros.actions import ComposableNodeContainer, Node
from launch_ros.descriptions import ComposableNode
```

vikasbz, 2021-11-23:
@AndyZe Can I please take this up?

AndyZe, 2021-11-23:
Please do!

INVINCIalokk, 2021-11-25:
sir i didn't understood this issue can you explain me......

sameergupta4873, 2023-01-30:
@AndyZe Can I work on this issue ?

sameergupta4873, 2023-02-03:
@AndyZe I had some free time, so I worked on the issue, please review the PR, also if @TanmayAgarwal123  had already started, I would love to assist him.