moveit/moveit2 #819: Hybrid planner often gets stuck

AndyZe, 2021-11-19:
Sometimes it plans around the obstacle perfectly, and executes the trajectory perfectly. Here's a perfect run:

https://user-images.githubusercontent.com/11284393/142681458-9f4775e2-58ef-490e-a70f-cc4060345f0a.mp4

Other times it gets stuck. It seems to be an issue with the local planner. Here is an example of that:

https://user-images.githubusercontent.com/11284393/142681077-e90f28d3-01bd-4bc0-a9e8-3c94b25a8b34.mp4

The terminal output when it gets stuck:

> [component_container_mt-1] [INFO] [1637350273.905547000] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
> [component_container_mt-1] [INFO] [1637350273.925019968] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'panda_arm'
> [component_container_mt-1] [INFO] [1637350273.928668550] [local_planner_component]: Received local planning goal request
> [hybrid_planning_demo_node-5] [INFO] [1637350273.928796992] [test_hybrid_planning_client]: Local goal accepted by server

JensVanhooydonck, 2021-11-20:
I'm also getting the following error in #813  when the global planner fails to find a solution:

```
[component_container_mt-1] [INFO] [1637409928.670715800] [hybrid_planning_manager]: Received goal request
[component_container_mt-1] [INFO] [1637409928.671226100] [global_planner_component]: Received global planning goal request
[component_container_mt-1] [WARN] [1637409928.671596200] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container_mt-1] [INFO] [1637409928.671768300] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'tmr_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container_mt-1] [WARN] [1637409930.683522700] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 2.011592 seconds
[component_container_mt-1] [INFO] [1637409930.683583100] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[component_container_mt-1] [ERROR] [1637409930.683598900] [moveit.ros_planning_interface.planning_component]: Could not compute plan successfully
[ERROR] [component_container_mt-1]: process has died [pid 31804, exit code -11, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container_mt --ros-args -r __node:=hybrid_planning_container -r __ns:=/'].
```

This crashes the hybrid_planning_container. New request are from then not possible anymore.

AndyZe, 2021-11-22:
@JensVanhooydonck this commit should fix your "planner fails" bug. It was a pretty simple fix. It would be nice if you could test this - I almost never see planning failures so it's hard for me to test.

https://github.com/ros-planning/moveit2/pull/813/commits/b9891f4163df9cb6cc5202b7cf7bcdd7ec101b25

Will work on the local planner getting stuck now.

JensVanhooydonck, 2021-11-23:
> @JensVanhooydonck this commit should fix your "planner fails" bug. It was a pretty simple fix. It would be nice if you could test this - I almost never see planning failures so it's hard for me to test.
> 
> [b9891f4](https://github.com/ros-planning/moveit2/commit/b9891f4163df9cb6cc5202b7cf7bcdd7ec101b25) (from PR #813)
> 
> Will work on the issue of the local planner getting stuck now.

Thanks for this! This indeed stops the container from crashing. However, after throwing the `Could not compute plan successfully`, new requests are ignored/not picked up.

I'm getting the following logs:
```
[component_container_mt-1] [WARN] [1637655549.515772700] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 2.008487 seconds
[component_container_mt-1] [INFO] [1637655549.515829900] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[component_container_mt-1] [ERROR] [1637655549.515838300] [moveit.ros_planning_interface.planning_component]: Could not compute plan successfully
[component_container_mt-1] [INFO] [1637655549.515856400] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'tmr_arm'
[component_container_mt-1] [ERROR] [1637655549.516277900] [moveit_robot_state.conversions]: Found empty JointState message
```

To fix this i've opened [PR #2](https://github.com/AndyZe/moveit2/pull/2) which fixes this for me.
The failing of the global planner is another issue outside of this scope.

AndyZe, 2021-11-23:
Merging it. Thanks @JensVanhooydonck!

closed 2021-11-23 by AndyZe