moveit/moveit2 #820: [foxy] std::bad_alloc, MoveGroup launched without ~default_planning_pipeline spec...,  'loadRobotModel'

JoNeckel, 2021-11-22:
### Description

I have a similar problem than in the fixed issue https://github.com/ros-planning/moveit2/issues/780#issue-1044731953

When I want to launch moveit for the UniversalRobots Driver, I get Errors and rviz tells me, that no planning library is loaded

### Your environment
* ROS Distro: [foxy] binary
* OS Version: e.g. Ubuntu 20.04

### Steps to reproduce
follow the instructions of https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/foxy
first launch controllers, then moveit

My Terminal output:
```
jo@myubuntu:~/workspace2/ros_ur_driver$ ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:="xxx.xxx" use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/jo/.ros/log/2021-11-22-14-00-45-135012-myubuntu-36904
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [36911]
[INFO] [mongo_wrapper_ros.py-2]: process started with pid [36913]
[INFO] [rviz2-3]: process started with pid [36915]
[INFO] [static_transform_publisher-4]: process started with pid [36917]
[static_transform_publisher-4] [INFO] [1637586047.218279834] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'base_link'
[move_group-1] [WARN] [1637586047.332429990] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1637586047.332522201] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Parsing robot urdf xml string.
[move_group-1] [INFO] [1637586047.344671497] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00286925 seconds
[move_group-1] [INFO] [1637586047.344750831] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[move_group-1] terminate called after throwing an instance of 'std::bad_alloc'
[move_group-1]   what():  std::bad_alloc
[mongo_wrapper_ros.py-2] [INFO] [1637586047.738997297] [mongodb]: Starting mongodb with db location /tmp/db listening on localhost:33829
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] MongoDB starting : pid=36956 port=33829 dbpath=/tmp/db 64-bit host=myubuntu
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] db version v3.6.8
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] allocator: tcmalloc
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] modules: none
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] build environment:
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten]     distarch: x86_64
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten]     target_arch: x86_64
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I CONTROL  [initandlisten] options: { net: { port: 33829 }, storage: { dbPath: "/tmp/db" } }
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I -        [initandlisten] Detected data files in /tmp/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I STORAGE  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I STORAGE  [initandlisten] ** WARNING: Using the XFS filesystem is strongly recommended with the WiredTiger storage engine
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I STORAGE  [initandlisten] **          See http://dochub.mongodb.org/core/prodnotes-filesystem
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:47.834+0100 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=7436M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
[ERROR] [move_group-1]: process has died [pid 36911, exit code -6, cmd '/home/jo/workspace2/ros_ur_driver/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_1uvxp47a --params-file /tmp/launch_params_wh4ihzet --params-file /tmp/launch_params_uqvhbica --params-file /tmp/launch_params_z38xeiyd --params-file /tmp/launch_params_z0f60w0g --params-file /tmp/launch_params_5fbdfx90 --params-file /tmp/launch_params_f7_e63_j'].
[rviz2-3] [INFO] [1637586048.748948902] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1637586048.749142702] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1637586048.828628151] [rviz2]: Stereo is NOT SUPPORTED
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:48.943+0100 I STORAGE  [initandlisten] WiredTiger message [1637586048:943694][36956:0x7fcdb35c5100], txn-recover: Main recovery loop: starting at 1/24832
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.051+0100 I STORAGE  [initandlisten] WiredTiger message [1637586049:51514][36956:0x7fcdb35c5100], txn-recover: Recovering log 1 through 2
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.144+0100 I STORAGE  [initandlisten] WiredTiger message [1637586049:144934][36956:0x7fcdb35c5100], txn-recover: Recovering log 2 through 2
[rviz2-3] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-3]          at line 253 in /opt/ros/foxy/include/class_loader/class_loader_core.hpp
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.212+0100 I STORAGE  [initandlisten] WiredTiger message [1637586049:212625][36956:0x7fcdb35c5100], txn-recover: Set global recovery timestamp: 0
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] ** WARNING: This server is bound to localhost.
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] **          Remote systems will be unable to connect to this server. 
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] **          Start the server with --bind_ip <address> to specify which IP 
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] **          addresses it should serve responses from, or with --bind_ip_all to
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] **          bind to all interfaces. If this behavior is desired, start the
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] **          server with --bind_ip 127.0.0.1 to disable this warning.
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.293+0100 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.325+0100 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '/tmp/db/diagnostic.data'
[mongo_wrapper_ros.py-2] 2021-11-22T14:00:49.328+0100 I NETWORK  [initandlisten] waiting for connections on port 33829
[rviz2-3] [ERROR] [1637586052.312925294] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] [INFO] [1637586052.520666487] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0105457 seconds
[... cut ...] 20804309] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[rviz2-3] [ERROR] [1637586052.566412057] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': std::bad_alloc
```

@JafarAbdi do you have any idea what else could be the problem or what I can try?
I had it working since you fixed it last time, but now out of nowhere it came again.. (I rebuilt my workspace once in a while but i don't know why this should be a problem)

JafarAbdi, 2021-11-22:
I did release a new version for foxy but it's not synced yet, do you mind trying the latest foxy branch https://github.com/ros-planning/moveit2/tree/foxy while removing the binaries? `sudo apt remove --purge ros-foxy-moveit*`

JoNeckel, 2021-11-22:
It works again, thank you!

closed 2021-11-22 by JoNeckel