moveit/moveit2 #826: Builderror moveit_ros_planning

NebuDa, 2021-11-24:
I have tried to build from Source on Ubuntu 20.04 with ROS2 Galactic and got the following errors:

```
[Processing: moveit_ros_planning]
--- stderr: moveit_ros_planning
/home/flo/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘void planning_scene_monitor::PlanningSceneMonitor::excludeWorldObjectFromOctree(const ObjectConstPtr&)’:
/home/flo/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:940:81: error: ‘using element_type = const struct collision_detection::World::Object’ {aka ‘const struct collision_detection::World::Object’} has no member named ‘global_shape_poses_’; did you mean ‘shape_poses_’?
  940 |       collision_body_shape_handles_[obj->id_].push_back(std::make_pair(h, &obj->global_shape_poses_[i]));
      |                                                                                 ^~~~~~~~~~~~~~~~~~~
      |                                                                                 shape_poses_
/home/flo/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘bool planning_scene_monitor::PlanningSceneMonitor::getShapeTransformCache(const string&, const rclcpp::Time&, occupancy_map_monitor::ShapeTransformCache&) const’:
/home/flo/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:1132:47: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
 1132 |             attached_body_shape_handle.first->getShapePosesInLinkFrame()[attached_body_shape_handle.second[k].second];
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/build.make:63: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/src/planning_scene_monitor.cpp.o] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[1]: *** [CMakeFiles/Makefile2:763: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/all] Fehler 2
make: *** [Makefile:141: all] Fehler 2
---
Failed   <<< moveit_ros_planning [1min 30s, exited with code 2]

Summary: 34 packages finished [9min 43s]
  1 package failed: moveit_ros_planning
  1 package had stderr output: moveit_ros_planning
  18 packages not processed

```

I have followed the tutorial exactly and did nothing different. https://moveit.ros.org/install-moveit2/source/

tylerjw, 2021-11-24:
Is there any chance you have moveit installed into /opt?  If you do it can cause build errors because it tries to link against moveit packages you have in /opt instead of those in your workspace.  You can remove them like this:

```
sudo apt purge "ros-${ROS_DISTRO}-moveit-*" 
```

If this is the issue it will be made better by ROS Humble.  OR is working on fixing this issue with how the ordering of linking is handled in ament/colcon.

NebuDa, 2021-11-25:
Thats it, Thank You very much!

closed 2021-11-25 by NebuDa

odellus, 2022-04-22:
I tried
```
sudo apt purge "ros-${ROS_DISTRO}-moveit-*" 
```
but when I get to this part of [the quickstart](https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html)
```
cd ~/ws_moveit2
colcon build --mixin release
```
it reinstalls the packages (I'm guessing to `/opt`) and I get the same error again. Should I open a new issue? I can't get the quickstart to work.