moveit/moveit2 #832: Problems when building from source

TsuunEvo, 2021-11-26:
### Description

Hello, i want to build moveit2 from source, but i get 2 errors, one is fixed with the hint in the console output, but i list it for completions sake.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source, main

### Steps to reproduce
Build moveit2 from source

### Expected behaviour
Build finishes ceanly

### Actual behaviour
I get the following console output when building package moveit_ros_planning

### Backtrace or Console output
```
--- stderr: moveit_ros_planning
/workspaces/moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘void planning_scene_monitor::PlanningSceneMonitor::excludeWorldObjectFromOctree(const ObjectConstPtr&)’:
/workspaces/moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:940:81: error: ‘using element_type = const struct collision_detection::World::Object’ {aka ‘const struct collision_detection::World::Object’} has no member named ‘global_shape_poses_’; did you mean ‘shape_poses_’?
  940 |       collision_body_shape_handles_[obj->id_].push_back(std::make_pair(h, &obj->global_shape_poses_[i]));
      |                                                                                 ^~~~~~~~~~~~~~~~~~~
      |                                                                                 shape_poses_
/workspaces/moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘bool planning_scene_monitor::PlanningSceneMonitor::getShapeTransformCache(const string&, const rclcpp::Time&, occupancy_map_monitor::ShapeTransformCache&) const’:
/workspaces/moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:1132:47: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
 1132 |             attached_body_shape_handle.first->getShapePosesInLinkFrame()[attached_body_shape_handle.second[k].second];
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/build.make:63: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/src/planning_scene_monitor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:763: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_ros_planning [3min 5s, exited with code 2]
```

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

JafarAbdi, 2021-11-26:
@TsuunEvo I think this should be fixed with https://github.com/ros-planning/moveit2/pull/830, do you mind testing it?

JafarAbdi, 2021-11-26:
This's a duplicate of https://github.com/ros-planning/moveit2/issues/826, please let me know if you still have the problem

closed 2021-11-26 by JafarAbdi