moveit/moveit2 #842: Test with address sanitizer

tylerjw, 2021-12-01:


tylerjw, 2021-12-01:
I've started working on this based on the instructions here: https://github.com/colcon/colcon-sanitizer-reports/blob/master/README.rst

To create the workspace these are the steps I've been using:
```
git clone https://github.com/ros-planning/moveit2.git src/moveit
vcs import src < src/moveit/moveit2.repos
rosinstall_generator moveit --rosdistro rolling --deps | vcs import src
rosinstall_generator ros2_control --rosdistro rolling --deps | vcs import src
rosinstall_generator ros2_controllers --rosdistro rolling --deps | vcs import src
rosinstall_generator control_toolbox --rosdistro rolling --deps | vcs import src
rosinstall_generator cv_bridge --rosdistro rolling --deps | vcs import src

rm -rf src/moveit
rm -rf src/moveit_resources/
git clone https://github.com/ros-planning/moveit2.git src/moveit
vcs import src < src/moveit/moveit2.repos
```

Then to build I've been using this command:
```
colcon build --build-base=build-asan --install-base=install-asan \
    --cmake-args -DOSRF_TESTING_TOOLS_CPP_DISABLE_MEMORY_TOOLS=ON \
                 -DINSTALL_EXAMPLES=OFF -DSECURITY=ON --no-warn-unused-cli \
                 -DCMAKE_BUILD_TYPE=Debug \
    --mixin asan-gcc ccache
```

Then I've tried testing moveit_servo like this:
```
colcon test --build-base=build-asan --install-base=install-asan \
                                                      --event-handlers sanitizer_report+ console_direct+ --packages-select moveit_servo
```

Which doesn't work because of some issue with linking asan.  I worked around that with:
```fish
set -x LD_PRELOAD /usr/lib/gcc/x86_64-linux-gnu/9/libasan.so
```

But then I ran into python losing its mind when it runs roslaunch (resulting in no tests actually running).  The next step is to either figure out why I needed the sledgehammer that is LD_PRELOAD or use some other environment variables to stop it from erroring in the python executable.

matemat13, 2024-04-29:
Hi! I know of two possible solutions:

1. Set the `LD_PRELOAD` environment variable in the launchfile only for the ROS node executable. This is possible in ROS1 with the `<env>` tag and I assume there is a similar way for ROS2 (don't know from the top of my head). I have tested it in ROS1 and it works.
2. It should be possible to define a suppression file for the address sanitizer as per [its documentation](https://github.com/google/sanitizers/wiki/AddressSanitizerLeakSanitizer#suppressions). I haven't tried this though.