moveit/moveit2 #855: Initialize Member Variables

tylerjw, 2021-12-01:


Faltu-Programmer, 2022-01-27:
hello,
i'm new to this Open Source world.
although i did complete my first contribution by writing my name on **firstcontributions/first-contributions**,

now i'm searching issues for contribute, and found this **Initialize Member Variables #855**
now i wondering, how can i contribute to solve this issue,
like what will be the steps?,
what should i change,  how should i change 

beiing honest I really don't understand what to do, can you @tylerjw please help me to learn this skill.....
thank you....

tylerjw, 2022-01-27:
The task is to basically follow the advice at the link in the issue and try to apply it to this repo in a PR.

Faltu-Programmer, 2022-01-27:
ok
thank you,
but i'm still confuse, on whice file or files i should make changes,
did i have to make change all the files on this repo by following advice at the link ?

Zandacrus, 2023-10-04:
Hello,
I am new to the open source world. I will contribute as much as I can.
Should I update the code for the main branch only ? Or for other branches as well ?

alexander-wang03, 2023-12-27:
Hello! I'm interested in contributing to this issue. Could you please provide any specific details or guidance on how to get started? Thanks!

shivangvijay, 2024-03-04:
https://github.com/ros2/rmw_zenoh/pull/121

shivangvijay, 2024-03-04:
https://github.com/ros2/rmw_zenoh/pull/122