moveit/moveit2 #857: Use .hpp extension for headers

tylerjw, 2021-12-01:


JorgePerC, 2023-09-25:
Hey, I would like to help, are there any specific instructions on how to do so?

TSNoble, 2024-11-16:
I'd be interested in picking this up if it's not being worked on?

TSNoble, 2024-11-18:
When #3133 is merged, the following items should finish off this issue:

[ ] Replace c library imports with c++ equivalents
[ ] Replace ros.h imports with rclcpp imports
[ ] Replace ROS1 msg imports with ROS2 msg imports
[ ] Replace fmtlib imports with std::format
[ ] Replace non-ROS library imports with hpp imports where possible
[ ] Get GENERATE_EXPORT_HEADER() macro to export hpps
[ ] Replace ikfast.h with ikfast.hpp (may be fixed by autogeneration of deprecated c-style headers)