moveit/moveit2 #868: Get rid of “new”

tylerjw, 2021-12-01:


Robotawi, 2022-05-31:
The `new` operator is still being used in some cases. 

Examples: 
- https://github.com/ros-planning/moveit2/blob/925866927b43b4f3ffa90f0ecb749019c3d1b578/moveit_core/distance_field/src/distance_field.cpp#L477 

- https://github.com/ros-planning/moveit2/blob/925866927b43b4f3ffa90f0ecb749019c3d1b578/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L1159 

Should those cases be considered for getting rid of `new`?
If yes, I’ll work on this for WMD 2022 @JafarAbdi

v4hn, 2022-05-31:
The examples you cite are obviously bad code either way.
The first one uses `new` multiple times(?) for a temporary structure (because it would be/was back in the days too much for the stack?).
If the function has to be reentrant that can make sense, but even then the style is horrible and should use one `unique_ptr` to a single allocated block instead.

The second one could probably be replaced by a local `JointConstraint` and a `emplace_back(std::move(...))` operation.
But it is bad design to begin with to have multiple lists that should be consistent all hold `shared_ptr` ownership to the same data.

It seems to me cleaning up the (/some of the) remaining `new` operators might be best done by reorganizing things in the datastructures themselves instead of replacing them by simple `make_*()` calls, so it might be better to look at a few of them only and think about how this can be avoided better by adapting the environment of the code.

TanmayAgarwal123, 2023-01-30:
I would love to work on this project, can you assign me
Regards,
Tanmay Agarwal