moveit/moveit2 #872: Never Use std::bind

tylerjw, 2021-12-01:


peterdavidfagan, 2022-04-08:
Hi @tylerjw, 

I'm happy to also pick up this issue as part of my GSoC application so feel free to assign this one to me as well. 

FYI @henningkayser.

v4hn, 2022-04-08:
I implemented that for moveit last week over the course of some hours: https://github.com/ros-planning/moveit/pull/3106
it's basically clang-tidy + assigning proper variable names.
However I also used the time to simplify and improve some code throughout the project.
With a million expected merge conflicts I would propose you go through the same steps again instead of meeting, but still take my pr as a reference and put all non-trivial commits please. :+1:

peterdavidfagan, 2022-04-08:
Awesome, thanks @v4hn it's great to have your pull request as a reference. I will go through the same steps as you have outlined above. 

I don't have the privileges to assign reviewers but it would be great to get your review of my pull request given your familiarity with the changes.

henningkayser, 2022-04-13:
@peterdavidfagan thanks for taking this on. Did you end up working on this already? Let us know if you need any help with this.

peterdavidfagan, 2022-04-13:
Hi @henningkayser, I have started working on this issue. I am aiming to open a pull request by this Friday/Saturday.

henningkayser, 2022-04-13:
> Hi @henningkayser, I have started working on this issue. I am aiming to open a pull request by this Friday/Saturday.

Awesome, thanks!

peterdavidfagan, 2022-04-25:
I just wished to provide a quick update on this issue, I am about 70% there with the changes. 

I am taking my time to go through this task as it is also a way for me to read through the codebase. I hope to submit a pr for this shortly, you may find the current status of changes on my fork [here](https://github.com/peterdavidfagan/moveit2/tree/remove_standard_bind). I also have a lot of local changes that still need to be pushed for the [banish bind commit](https://github.com/ros-planning/moveit/pull/3106/commits/a2911c80c28958c1fce8fb52333d770248c4ec05).

closed 2022-05-10 by henningkayser