moveit/moveit2 #881: Use the correct integer type for standard library features

tylerjw, 2021-12-01:
https://github.com/cpp-best-practices/cppbestpractices/blob/master/03-Style.md#use-the-correct-integer-type-for-standard-library-features

Shanki5, 2022-02-14:
@tylerjw I would like to take up this issue if nobody is working on this. Can you also elaborate a bit on the issue?

tylerjw, 2022-02-14:
Please just follow the advice that is in that link and open a pull request with the changes.

henningkayser, 2023-01-26:
aren't we already doing this consistently?

TaMaN2031A, 2023-04-15:
> Please just follow the advice that is in that link and open a pull request with the changes.

It would be very helpful if you can mention any example so that we could work on it.
I wonder if it is the size_t that needs to be replaced with an integer declaration.
Something like this:
`
[image]
`
In general, what is the logic to be followed in fixing this issue?