moveit/moveit2 #885: Fix circular dependency with moveit_resources (PILZ/PRBT)

henningkayser, 2021-12-03:
Tracking issue for the MoveIt 2 side of https://github.com/ros-planning/moveit/issues/2947.

* The PILZ planner should use a supported robot rather than PRBT
* The circular dependency with moveit_resources should be fixed

v4hn, 2021-12-23:
reopening because the "fix" PR actually states itself that this is a temporary workaround and does not work towards the solution...

tylerjw, 2021-12-23:
It is a solution of sorts.  There is no more circular dependency.  When we have time I want to migrate the tests to a different robot.  Until we figure out how to have deterministic IK without depending on moveit_core we won't get away from having the test config for pilz in moveit itself.  Even if we switch robots for the tests.

v4hn, 2021-12-23:
> There is no more circular dependency.

But that's the symptom, not the underlying issue (no deterministic IK+robot config in MoveIt&resources).
Now the symptom is gone and people have even less drive to fix the real issue...

> Until we figure out how to have deterministic IK without depending on moveit_core we won't get away from having the test config for pilz in moveit itself. Even if we switch robots for the tests.

I don't follow. As Robert explained nicely in [the original issue](https://github.com/ros-planning/moveit/issues/2947) adding the opw plugin would be a nice community contribution and would directly allow to make fanuc a deterministic IK config in moveit_resources.

tylerjw, 2021-12-23:
> I don't follow. As Robert explained nicely in [the original issue](https://github.com/ros-planning/moveit/issues/2947) adding the opw plugin would be a nice community contribution and would directly allow to make fanuc a deterministic IK config in moveit_resources.

That would be a really nice contribution.  This "solution" does not prevent someone from doing that.  As I stated in the other issue, I don't care about this more than other things.  The primary reason I don't care about this is I don't think I am the right person to try to tackle this problem, and instead I should focus on improving things that are more in my domain.

henningkayser, 2021-12-23:
I agree that we should get this fixed, supporting the opw plugin efforts makes a lot of sense to me. I'm fine with keeping this issue open until it's really fixed. The reason we picked this solution is to unblock all the other work that's going on.

v4hn, 2021-12-23:
> The reason we picked this solution is to unblock all the other work that's going on.

I'm still wondering why this blocked anything, but maybe I just did not read the correct issues...
The only thing I can find is https://github.com/ros-planning/moveit2/commit/0b908a172d778519766486807684232e3f9bbe62 ,
which is so much less invasive than what you did now...

PR #909 (2021-12-22): Pilz test_resources
- Adding RPBT config
- prbt_support: drop all test code
- Port prbt packages to ROS 2
- Move PRBT into test_configs directory

### Description

This fixes the circular dependency with moveit_resources.  This is a temporary solution until we are able to test the Pilz planners with a different robot.

I preserved the commit history of these by cherry-picking from moveit_resources.

This depends on this PR: https://github.com/ros-planning/moveit_resources/pull/106

This fixes #885 

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://move [... cut]