moveit/moveit2 #892: Consistent parameter loading for the different motion planning pipelines

AndyZe, 2021-12-06:
hybrid_planning and moveit_task_constructor are two components of MoveIt that are currently supposed to work with interchangeable planning pipelines, e.g. CHOMP and OMPL.

This gets tricky in ROS2 because CHOMP and OMPL have different parameters and we don't know what they will be ahead of time. So, hard-coded versions of `declare_parameter` are not really possible.

One way to get around this in some cases is to set `node_options.automatically_declare_parameters_from_overrides(true);`. Then, whatever you have in the yaml file will be automatically declared. I think this is a decent solution but it doesn't do any type or error catching.

`node_options.automatically_declare_parameters_from_overrides(true);` does not work for plugins because the constructor is passed a const node_options argument, e.g.

`GlobalPlannerComponent::GlobalPlannerComponent(const rclcpp::NodeOptions& options)`

^ So we cannot change the node options for this plugin.

@sjahr has suggested that OMPL can read its own parameters in the OMPL interface class. That seems like a good idea to me.

Another option is to look at how moveit_cpp does it. Does it have a good general solution that could be used in other places?

tylerjw, 2021-12-06:
If each component/plugin that needs parameters declares and loads its own parameters then it should work.  Is that what you mean when you say OMPL could load it's own parameters in the OMPL interface?

tylerjw, 2021-12-06:
It is probably also worth reviewing what ros2_control is doing for this as they have a similar situation where they don't know what parameters they need until they load plugins.