moveit/moveit2 #897: Enable CppCheck in MoveIt2

Abishalini, 2021-12-08:
CppCheck is a static analysis tool that can detect undefined behavior like 
*    Dead pointers
*    Division by zero
*    Integer overflows
*    Invalid bit shift operands
*    Invalid conversions
*    Invalid usage of STL
*    Memory management
*    Null pointer dereferences
*    Out of bounds checking
*    Uninitialized variables
*    Writing const data

Currently, CppCheck is disable for this codebase by adding this line in CMakeLists
```
set(ament_cmake_cppcheck_FOUND TRUE)
```

It would be nice to add this back. This task can be done incrementally by enabling CppCheck for each library in `moveit_ros` and `moveit_core` separately.

v0nser, 2022-01-28:
Hey! I would like to do this one.

AKACHI1, 2022-02-05:
Hello, I am new here and want to contribute in your project I would like to ask what i should learn to contribute here and please also tell me some resources ....

Abishalini, 2022-02-08:
A good starting place is our [tutorials website](https://moveit.picknik.ai/galactic/index.html).
The tutorials are continuously being improved  and if you run into any issues, feel free to open an issue in the [tutorials repository](https://github.com/ros-planning/moveit2_tutorials).

One of my colleague also posted a [few YouTube videos](https://picknik.ai/series/Getting-Started-With-Moveit2) that walks you through the tutorials.

As for this issue, CppCheck is disabled for multiple components of MoveIt2. Chose any component (like `moveit_core`) and delete the following line from the `CMakeLists.txt`
```
set(ament_cmake_cppcheck_FOUND TRUE)
```
Building your workspace after this change should show you errors that CppCheck detects in that package. If you fix all the errors, create a pull request with your changes.

DolceParadise, 2022-02-15:
Hi! I'd like to work on this issue if it isn't already allotted to someone else

henningkayser, 2022-06-02:
cppcheck has been enabled with #1224

closed 2022-06-02 by henningkayser