moveit/moveit2 #919: Foxy build fails due undefined reference to `boost::iostreams::[...]

slabua, 2021-12-14:
### Description

Foxy build fails due to /usr/bin/ld: libmoveit_distance_field.so.2.2.2: undefined reference to `boost::iostreams::[...]'

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04.3
* Source build
* foxy branch

### Steps to reproduce
Following the build instruction from https://moveit.ros.org/install-moveit2/source/

### Expected behaviour
Build should complete as expected.

### Actual behaviour
Build stops with the following console log:

### Backtrace or Console output

https://gist.github.com/slabua/f33f997f2a202a4c604570a489bb14b9

vatanaksoytezer, 2021-12-14:
Thanks for reporting @slabua! Seems https://github.com/ros-planning/geometric_shapes/pull/211 broke it. Backporting #900 should fix it. Sending a PR now.

vatanaksoytezer, 2021-12-14:
https://github.com/ros-planning/moveit2/pull/920

slabua, 2021-12-14:
@vatanaksoytezer thanks! I will try it tomorrow.

RoboticsYY, 2021-12-14:
Reproduced the issue on a fresh build today. Backporting `geometric_shapes` fixed it.

slabua, 2021-12-15:
@vatanaksoytezer thanks, it passed that part successfully but I am having another problem related to `/usr/bin/ld: /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0: error adding symbols: DSO missing from command line`, I will open a separate issue

closed 2021-12-15 by slabua