moveit/moveit2 #923: Foxy libboost_serialization.so.1.71.0: error adding symbols: DSO missing

slabua, 2021-12-15:
### Description

Foxy build fails due to libboost_serialization.so.1.71.0: error adding symbols: DSO missing from command line

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04.3
* Source build
* foxy branch

### Steps to reproduce
Following the build instruction from https://moveit.ros.org/install-moveit2/source/

### Expected behaviour
Build should complete as expected.

### Actual behaviour
Build stops with the following console log:

### Backtrace or Console output

```
[Processing: moveit_planners_ompl, moveit_ros_benchmarks, moveit_ros_move_group]
--- stderr: moveit_planners_ompl
/usr/bin/ld: CMakeFiles/test_planning_context_manager.dir/test/test_planning_context_manager.cpp.o: undefined reference to symbol '_ZN5boost7archive6detail22archive_serializer_mapINS0_15binary_oarchiveEE6insertEPKNS1_16basic_serializerE'
/usr/bin/ld: /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [ompl_interface/CMakeFiles/test_planning_context_manager.dir/build.make:326: ompl_interface/test_planning_context_manager] Error 1
make[1]: *** [CMakeFiles/Makefile2:386: ompl_interface/CMakeFiles/test_planning_context_manager.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_planners_ompl [4min 21s, exited with code 2]
```

vatanaksoytezer, 2021-12-15:
Docker build seems to reproduce this issues as well here: https://github.com/ros-planning/moveit2/runs/4523535433?check_suite_focus=true

I'll take a look at it now and try to send a fix.

vatanaksoytezer, 2021-12-15:
https://github.com/ros-planning/moveit2/pull/925 should fix can you confirm @slabua?

parkbytes, 2021-12-15:
Was getting same error as reproduced in docker  and #925 fixed it for me.  Thanks!

vatanaksoytezer, 2021-12-15:
Closed via #925. Feel free to re-open this issue or a open a new one if you are having any troubles.

closed 2021-12-15 by vatanaksoytezer

slabua, 2021-12-16:
@vatanaksoytezer thanks! Fixed for me too.

PR #925 (2021-12-15): [foxy] Link boost serializaiton
Closes #923. Somehow CI wasn't able to catch this (I'm assuming it was caching that part), but docker action caught this as well (https://github.com/ros-planning/moveit2/runs/4523535433?check_suite_focus=true). This PR just adds a one linn to properly link boost_serialization library in `ompl_interface`. Tested by building the docker locally. I am not sure why we haven't seen this in main though.