moveit/moveit2 #927: Controller interface for servo (vibe)

tylerjw, 2021-12-16:
The xbox controller interface already exists here: https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/src/teleop_demo/joystick_servo_example.cpp

I would like a similar system controller component node for the htc vipe controller.

I would like an image like this one showing what each button (and joystick) does:
[image]

This should go in a tutorial for how you use moveit_servo with this controller.  An example launchfile that opens rviz with the panda in simulation using this controller should be included.

AndyZe, 2021-12-16:
In ROS1 we used the teleop-tools package rather than creating an executable for each controller type. It lets you set up a new controller very easily, with just a config file.

See https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/spacenav_teleop_tools.launch

I don't want 8 executables for 8 different controller types -- that would be a lot of maintenance overhead.

Do we really want a launch file for every different controller type? Again, that is a lot of maintenance.

AndyZe, 2021-12-16:
Thinking about it more, I'm quite sure there should be one launch file for the different teleop_tools controller types. An argument to switch between them would be great.

tylerjw, 2021-12-16:
I am all for not copy-pasted code we have to maintain.  If there is a more abstract solution that lets us implement a generic controller interface with a config for each I am all for that.  @ParthasarathyBana do you want to look into this?

What I was trying to express with these is it would be really helpful if there was a really straightforward way for users to use many different controllers with good documentation it would be a welcome addition.

tylerjw, 2021-12-16:
The launch file for this could exist in the moveit2_tutorials package if necessary but it would be really nice if someone could easily switch between controllers with standardized control interfaces that are well defined.