moveit/moveit2 #947: Segfault in Servo "ExternalCollision" test

henningkayser, 2021-12-30:
### Description

Failing CI because of Segfault in Servo test.

See https://github.com/ros-planning/moveit2/runs/4665772418?check_suite_focus=true#step:9:12904
```
12904    [test_servo_collision-6] [INFO] [1640861996.843502159] [moveit_servo.servo_launch_test_common.hpp]: Setup time: 0.500997
12905    [servo_node_main-2] [WARN] [1640861997.022630541] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12906    [servo_node_main-2] [WARN] [1640861998.024138407] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12907    [servo_node_main-2] [WARN] [1640861999.035353542] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12908    [servo_node_main-2] [WARN] [1640862000.046106274] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12909    [servo_node_main-2] [WARN] [1640862001.049553558] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12910    [servo_node_main-2] [WARN] [1640862002.081152181] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12911    [servo_node_main-2] [WARN] [1640862003.081609293] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12912    [servo_node_main-2] [WARN] [1640862004.082821410] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12913    [servo_node_main-2] [WARN] [1640862005.085372836] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12914    [servo_node_main-2] [WARN] [1640862006.086400552] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing virtual_joint
12915    [test_servo_collision-6] [INFO] [1640862006.897628675] [moveit_servo.servo_launch_test_common.hpp]: Wait for start servo: 10.0541
12916    [test_servo_collision-6] [INFO] [1640862006.947787417] [moveit_servo.servo_launch_test_common.hpp]: Wait for status num increasing: 0.0501102
12917    [servo_node_main-2] Stack trace (most recent call last) in thread 21167:
12918    [servo_node_main-2] #23   Object "", at 0xffffffffffffffff, in 
12919    [servo_node_main-2] #22   Object "/usr/lib/x86_64-linux-gnu/libc-2.31.so", at 0x7efd39d75292, in __clone
12920    [servo_node_main-2] #21   Object "/usr/lib/x86_64-linux-gnu/libpthread-2.31.so", at 0x7efd398ad608, in start_thread
12921    [servo_node_main-2] #20   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7efd39f36de3, in 
12922    [servo_node_main-2] #19   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7efd3a393d9b, in rclcpp::Executors::SingleThreadedExecutor::spin()
12923    [servo_node_main-2] #18   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7efd3a38dd90, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
12924    [servo_node_main-2] #17   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7efd3a38cb63, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
12925    [servo_node_main-2] #16   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7efd3a38c060, in 
12926    [servo_node_main-2] #15   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7efd3a42b6b1, in rclcpp::SubscriptionBase::take_type_erased(void*, rclcpp::MessageInfo&)
12927    [servo_node_main-2] #14   Object "/opt/ros/rolling/lib/librcl.so", at 0x7efd39bb3b1a, in rcl_take
12928    [servo_node_main-2] #13   Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd3585ba04, in 
12929    [servo_node_main-2] #12   Object "/opt/ros/rolling/lib/x86_64-linux-gnu/libddsc.so.0.8.0", at 0x7efd357c784a, in dds_take
12930    [servo_node_main-2] #11   Object "/opt/ros/rolling/lib/x86_64-linux-gnu/libddsc.so.0.8.0", at 0x7efd357c700c, in 
12931    [servo_node_main-2] #10   Object "/opt/ros/rolling/lib/x86_64-linux-gnu/libddsc.so.0.8.0", at 0x7efd357bae97, in 
12932    [servo_node_main-2] #9    Object "/opt/ros/rolling/lib/x86_64-linux-gnu/libddsc.so.0.8.0", at 0x7efd357badd6, in 
12933    [servo_node_main-2] #8    Object "/opt/ros/rolling/lib/x86_64-linux-gnu/libddsc.so.0.8.0", at 0x7efd357ba76a, in 
12934    [servo_node_main-2] #7    Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd35895bab, in 
12935    [servo_node_main-2] #6    Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd3589a5ec, in 
12936    [servo_node_main-2] #5    Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd3587d19a, in 
12937    [servo_node_main-2] #4    Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd3587c5b8, in 
12938    [servo_node_main-2] #3    Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd3587c5b8, in 
12939    [servo_node_main-2] #2    Object "/opt/ros/rolling/lib/librmw_cyclonedds_cpp.so", at 0x7efd3587cf68, in 
12940    [servo_node_main-2] #1    Object "/opt/ros/rolling/lib/libshape_msgs__rosidl_typesupport_introspection_cpp.so", at 0x7efd37de5aec, in shape_msgs::msg::rosidl_typesupport_introspection_cpp::resize_function__SolidPrimitive__polygon(void*, unsigned long)
12941    [servo_node_main-2] #0    Object "/usr/lib/x86_64-linux-gnu/libc-2.31.so", at 0x7efd39cf0870, in cfree
12942    [servo_node_main-2] Segmentation fault (Signal sent by the kernel [(nil)])
12943 Error: ROR] [servo_node_main-2]: process has died [pid 21126, exit code -11, cmd '/home/runner/work/moveit2/moveit2/.work/target_ws/install/moveit_servo/lib/moveit_servo/servo_node_main --ros-args --params-file /tmp/launch_params_o5nlnxdx --params-file /tmp/launch_params_a06kqxro --params-file /tmp/launch_params_37g5gzid --params-file /tmp/launch_params_wzz6sg7c'].
12944    [test_servo_collision-6] /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/moveit_servo/test/test_servo_collision.cpp:138: Failure
12945    [test_servo_collision-6] Expected: (iterations) < (test_parameters_->timeout_iterations), actual: 201 vs 200
12946    [test_servo_collision-6] [INFO] [1640862016.949212966] [moveit_servo.servo_launch_test_common.hpp]: Wait for collision: 10.0002
12947    [test_servo_collision-6] [  FAILED  ] ServoFixture.ExternalCollision (20816 ms)
12948    [test_servo_collision-6] [----------] 2 tests from ServoFixture (31327 ms total)
12949    [test_servo_collision-6] 
12950    [test_servo_collision-6] [----------] Global test environment tear-down
12951    [test_servo_collision-6] [==========] 2 tests from 1 test suite ran. (31328 ms total)
12952    [test_servo_collision-6] [  PASSED  ] 1 test.
12953    [test_servo_collision-6] [  FAILED  ] 1 test, listed below:
12954    [test_servo_collision-6] [  FAILED  ] ServoFixture.ExternalCollision
12955    [test_servo_collision-6] 
12956    [test_servo_collision-6]  1 FAILED TEST
12957 Error: ROR] [test_servo_collision-6]: process has died [pid 21191, exit code 1, cmd '/home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_servo/test_servo_collision --ros-args --params-file /tmp/launch_params_f09xwiac'].
```

AndyZe, 2021-12-31:
I see in the log, this has something to do with shape_msgs.

`shape_msgs::msg::rosidl_typesupport_introspection_cpp::resize_function__SolidPrimitive__polygon(void*, unsigned long)`

Wonder if it's related to this commit:  https://github.com/ros2/common_interfaces/commit/90a4602f0a7c1792de65e1416fd2d5f8822bb562

AndyZe, 2021-12-31:
Indeed, it looks like v4.0.0 of `common_interfaces` does not work on Rolling. Checking out an earlier commit fixes it. Not sure what can be done about this.

https://github.com/ros-planning/moveit2/pull/950

tylerjw, 2022-01-02:
I think this is a side-afect of the same thing that caused this: https://github.com/ros-planning/moveit2/issues/783

Updating the repos file (the other PR I merged) caused us to re-build our upstream workspace cache which re-linked with the new libraries in ros2 and made this issue go away.  I think this issue can be closed as we have a plan for what to do about this.  @JafarAbdi is going to add a string representing the datetime that the rolling repo was updated to invalidate our workspace caches in CI.

tylerjw, 2022-01-03:
Closing as this seems to be fixed by #952

closed 2022-01-03 by tylerjw