moveit/moveit2 #963: Failed   <<< moveit_visual_tools [1min 43s, exited with code 2]

Ghotic190, 2022-01-04:
Hello i`m trying to install moveit2 on raspi but can`t go throw this i`m following this guide [--->](https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html#install-ros-2-and-colcon)

when i try :

```
cd ~/ws_moveit2
colcon build --mixin release
```
i got this error

```
root@ros:~/ws_moveit2# colcon build --mixin release
[0.260s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'moveit_resources_fanuc_description' is in: /root/ws_moveit2/install/moveit_resources_fanuc_description
	'moveit_resources_panda_description' is in: /root/ws_moveit2/install/moveit_resources_panda_description
	'moveit_resources_pr2_description' is in: /root/ws_moveit2/install/moveit_resources_pr2_description
	'rviz_visual_tools' is in: /root/ws_moveit2/install/rviz_visual_tools
	'moveit_resources_fanuc_moveit_config' is in: /root/ws_moveit2/install/moveit_resources_fanuc_moveit_config
	'moveit_resources_panda_moveit_config' is in: /root/ws_moveit2/install/moveit_resources_panda_moveit_config
	'moveit_visual_tools' is in: /root/ws_moveit2/install/moveit_visual_tools
	'moveit_resources' is in: /root/ws_moveit2/install/moveit_resources
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding moveit_resources moveit_resources_fanuc_description moveit_resources_fanuc_moveit_config moveit_resources_panda_description moveit_resources_panda_moveit_config moveit_resources_pr2_description moveit_visual_tools rviz_visual_tools

This may be promoted to an error in a future release of colcon-core.
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> rviz_visual_tools
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [0.36s]                                  
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_description [0.43s]                                  
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [0.50s]                     
Finished <<< moveit_resources_fanuc_moveit_config [0.30s]                 
Finished <<< moveit_resources_panda_moveit_config [0.25s]                 
Starting >>> moveit_resources
Finished <<< moveit_resources [0.17s]                                                                  
Finished <<< rviz_visual_tools [0.98s]                  
Starting >>> moveit_visual_tools
--- stderr: moveit_visual_tools                               
In file included from /opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:21,
                 from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/src/moveit_visual_tools.cpp:32:
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const Edges&, std::ostream&, size_t)’:
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:36:9: error: ‘value_to_yaml’ was not declared in this scope
   36 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:56:9: error: ‘value_to_yaml’ was not declared in this scope
   56 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
In file included from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/src/moveit_visual_tools.cpp:32:
/opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const GeometryGraph&, std::ostream&, size_t)’:
/opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:36:45: error: no matching function for call to ‘to_yaml(const _header_type&, std::ostream&, size_t)’
   36 |     to_yaml(msg.header, out, indentation + 2);
      |                                             ^
In file included from /opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:21,
                 from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/src/moveit_visual_tools.cpp:32:
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:18:13: note: candidate: ‘void rosidl_generator_traits::to_yaml(const Edges&, std::ostream&, size_t)’
   18 | inline void to_yaml(
      |             ^~~~~~~
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:19:34: note:   no known conversion for argument 1 from ‘const _header_type’ {aka ‘const std_msgs::msg::Header_<std::allocator<void> >’} to ‘const Edges&’ {aka ‘const graph_msgs::msg::Edges_<std::allocator<void> >&’}
[... 176 lines cut ...]
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:63:20: note:   candidate expects 1 argument, 3 provided
In file included from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/imarker_robot_state.h:43,
                 from /root/ws_moveit2/src/moveit_visual_tools/src/imarker_end_effector.cpp:47:
/opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:26:13: note: candidate: ‘void rosidl_generator_traits::to_yaml(const GeometryGraph&, std::ostream&, size_t)’
   26 | inline void to_yaml(
      |             ^~~~~~~
/opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:27:42: note:   no known conversion for argument 1 from ‘geometry_msgs::msg::Point_<std::allocator<void> >’ to ‘const GeometryGraph&’ {aka ‘const graph_msgs::msg::GeometryGraph_<std::allocator<void> >&’}
   27 |   const graph_msgs::msg::GeometryGraph & msg,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
make[2]: *** [CMakeFiles/moveit_visual_tools.dir/build.make:76: CMakeFiles/moveit_visual_tools.dir/src/imarker_robot_state.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/moveit_visual_tools.dir/build.make:89: CMakeFiles/moveit_visual_tools.dir/src/imarker_end_effector.cpp.o] Error 1
make[2]: *** [CMakeFiles/moveit_visual_tools.dir/build.make:63: CMakeFiles/moveit_visual_tools.dir/src/moveit_visual_tools.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/moveit_visual_tools.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_visual_tools [1min 43s, exited with code 2]
                                      
Summary: 7 packages finished [1min 44s]
  1 package failed: moveit_visual_tools
  1 package had stderr output: moveit_visual_tools
  1 package not processed
```

Abishalini, 2022-01-05:
The errors seem to be coming from ROS2 message generation code. If you installed ROS2 via debian packages, try updating your ROS2 packages by `apt-get update && apt-get upgrade`

Ghotic190, 2022-01-05:
> The errors seem to be coming from ROS2 message generation code. If you installed ROS2 via debian packages, try updating your ROS2 packages by `apt-get update && apt-get upgrade`

Still same output 

`root@ros:~# cd ~/ws_moveit2
root@ros:~/ws_moveit2# colcon build --mixin release
[7.573s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'moveit_resources_fanuc_description' is in: /root/ws_moveit2/install/moveit_resources_fanuc_description
	'moveit_resources_panda_description' is in: /root/ws_moveit2/install/moveit_resources_panda_description
	'moveit_resources_pr2_description' is in: /root/ws_moveit2/install/moveit_resources_pr2_description
	'rviz_visual_tools' is in: /root/ws_moveit2/install/rviz_visual_tools
	'moveit_resources_fanuc_moveit_config' is in: /root/ws_moveit2/install/moveit_resources_fanuc_moveit_config
	'moveit_resources_panda_moveit_config' is in: /root/ws_moveit2/install/moveit_resources_panda_moveit_config
	'moveit_visual_tools' is in: /root/ws_moveit2/install/moveit_visual_tools
	'moveit_resources' is in: /root/ws_moveit2/install/moveit_resources
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding moveit_resources moveit_resources_fanuc_description moveit_resources_fanuc_moveit_config moveit_resources_panda_description moveit_resources_panda_moveit_config moveit_resources_pr2_description moveit_visual_tools rviz_visual_tools

This may be promoted to an error in a future release of colcon-core.
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description                  
Starting >>> rviz_visual_tools                                                   
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [4.04s]                                  
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_pr2_description [4.02s]                                    
Finished <<< moveit_resources_panda_description [4.32s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [1.04s]                 
Finished <<< moveit_resources_fanuc_moveit_config [1.44s]                 
Starting >>> moveit_resources
Finished <<< moveit_resources [1.09s]                                                                 
Finished <<< rviz_visual_tools [7.99s]                      
Starting >>> moveit_visual_tools
[Processing: moveit_visual_tools]                             
--- stderr: moveit_visual_tools                               
In file included from /opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:21,
                 from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/imarker_robot_state.h:43,
                 from /root/ws_moveit2/src/moveit_visual_tools/src/imarker_end_effector.cpp:47:
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const Edges&, std::ostream&, size_t)’:
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:36:9: error: ‘value_to_yaml’ was not declared in this scope
   36 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
/opt/ros/galactic/include/graph_msgs/msg/detail/edges__traits.hpp:56:9: error: ‘value_to_yaml’ was not declared in this scope
   56 |         value_to_yaml(item, out);
      |         ^~~~~~~~~~~~~
In file included from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/imarker_robot_state.h:43,
                 from /root/ws_moveit2/src/moveit_visual_tools/src/imarker_end_effector.cpp:47:
/opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const GeometryGraph&, std::ostream&, size_t)’:
/opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:36:45: error: no matching function for call to ‘to_yaml(const _header_type&, std::ostream&, size_t)’
   36 |     to_yaml(msg.header, out, indentation + 2);
      |                                             ^
In file included from /opt/ros/galactic/include/graph_msgs/msg/detail/geometry_graph__traits.hpp:21,
                 from /opt/ros/galactic/include/graph_msgs/msg/geometry_graph.hpp:9,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:52,
                 from /root/ws_moveit2/src/moveit_visual_tools/include/moveit_visual_tools/imarker_robot_state.h:43,
[... cut ...]
make[2]: *** [CMakeFiles/moveit_visual_tools.dir/build.make:89: CMakeFiles/moveit_visual_tools.dir/src/imarker_end_effector.cpp.o] Error 1
make[2]: *** [CMakeFiles/moveit_visual_tools.dir/build.make:63: CMakeFiles/moveit_visual_tools.dir/src/moveit_visual_tools.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/moveit_visual_tools.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_visual_tools [40.2s, exited with code 2]

Summary: 7 packages finished [51.7s]
  1 package failed: moveit_visual_tools
  1 package had stderr output: moveit_visual_tools
  1 package not processed
`

vatanaksoytezer, 2022-01-17:
`moveit_visual_tools` seem to work fine for CI and local. There might be some problem with your environment setup as @Abishalini suggested. In any case I am closing this issue, because this is not related with moveit2. Please open an issue in [moveit_visual_tools](https://github.com/ros-planning/moveit_visual_tools) repository if you are still having troubles.

closed 2022-01-17 by vatanaksoytezer

Ghotic190, 2022-01-17:
I solve this issue by reinsaling ros and moveit