moveit/moveit2 #966: MoveGroupInterface /robot_description not found

Splinter1984, 2022-01-06:
**Environment**

- ROS Distro: Foxy
- OS Version: e.g. Ubuntu 20.04
- Debian packages install

**Description**

how to pass robot_description if i run MoveGroupInterface as part of Plansys2 BTAction?

I am trying to use MoveGroupInterface as part of plansys2 and I am getting error

```bash
[bt_action_node-2] [INFO] [1641492464.774760696] [moveit_rdf_loader.rdf_loader]: Robot model parameter not found! Did you remap '/robot_description'?
[bt_action_node-2] [FATAL] [1641492464.774817909] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
```

i am just trying to initialize MoveGroupInterface in main tick start method
```cpp

    rclcpp::NodeOptions node_options;
    node_options.automatically_declare_parameters_from_overrides(true);
    auto move_group_node = rclcpp::Node::make_shared("move_group_interface_tutorial", node_options);

    //Setup MoveIt interface for planning
    moveit::planning_interface::MoveGroupInterface::Options move_group_options("robot_group", "/robot_description");
    moveit::planning_interface::MoveGroupInterface move_group_interface(move_group_node, move_group_options);
```

and run the node together with plansys2

```python
    move_1 = Node(
    package='plansys2_bt_actions',
    executable='bt_action_node',
    name='move',
    namespace=namespace,
    output='screen',
    parameters=[
      pkg_dir + '/config/params.yaml',
      {
        'action_name': 'move',
        'bt_xml_file': pkg_dir + '/behavior_trees_xml/move.xml'
      }
    ])
```

JafarAbdi, 2022-01-07:
You need to load the robot description parameter to the node [see](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/move_group_interface/launch/move_group_interface_tutorial.launch.py#L54)

closed 2022-01-07 by JafarAbdi

Splinter1984, 2022-01-08:
thanks for your answer, but the fact is that I'm trying to run MoveGroupInterface inside BTAction and, accordingly, I have no idea how easy it is to pass parameters about 'robot_description' from launch file to MoveGroupInterface inside BTAction node.