moveit/moveit2 #968: Hybrid Planning test silently fails, broken controller setup

henningkayser, 2022-01-06:
### Description

The hybrid planning test produces an `RCLError` exception which kills the hybrid planning node.
The test itself doesn't fail but it claims to have 0 test results.
The reason is probably that the controller spawner fails ungratefully since the `panda_joint_group_position_controller` type is not available.

The exception was found by @vatanaksoytezer [here](https://github.com/ros-planning/moveit2/pull/953#issuecomment-1004293698). At first I thought it was an issue with the galactic branch, but this also happens with the galactic and rolling jobs on `main`.

### Backtrace or Console output
```
2022-01-06T16:30:02.5146837Z 1: [ros2_control_node-6] [INFO] [1641486596.458964111] [controller_manager]: Loading controller 'joint_state_broadcaster'
2022-01-06T16:30:02.5148088Z 1: [ros2_control_node-6] [INFO] [1641486596.474669114] [controller_manager]: Loading controller 'panda_joint_group_position_controller'
2022-01-06T16:30:02.5149396Z 1: [ros2_control_node-6] [ERROR] [1641486596.474726014] [controller_manager]: Available classes:
2022-01-06T16:30:02.5150459Z 1: [ros2_control_node-6] [ERROR] [1641486596.474738014] [controller_manager]: controller_manager/test_controller
2022-01-06T16:30:02.5151618Z 1: [ros2_control_node-6] [ERROR] [1641486596.474745814] [controller_manager]: controller_manager/test_controller_failed_init
2022-01-06T16:30:02.5152836Z 1: [ros2_control_node-6] [ERROR] [1641486596.474753514] [controller_manager]: controller_manager/test_controller_with_interfaces
2022-01-06T16:30:02.5154104Z 1: [ros2_control_node-6] [ERROR] [1641486596.474761315] [controller_manager]: effort_controllers/GripperActionController
2022-01-06T16:30:02.5155420Z 1: [ros2_control_node-6] [ERROR] [1641486596.474769215] [controller_manager]: joint_state_broadcaster/JointStateBroadcaster
2022-01-06T16:30:02.5159399Z 1: [ros2_control_node-6] [ERROR] [1641486596.474776315] [controller_manager]: joint_trajectory_controller/JointTrajectoryController
2022-01-06T16:30:02.5160908Z 1: [ros2_control_node-6] [ERROR] [1641486596.474783915] [controller_manager]: position_controllers/GripperActionController
2022-01-06T16:30:02.5162288Z 1: [ros2_control_node-6] [ERROR] [1641486596.474791015] [controller_manager]: Loader for controller 'panda_joint_group_position_controller' not found.
2022-01-06T16:30:02.5163767Z 1: [ros2 run controller_manager spawner joint_state_broadcaster-7] [INFO] [1641486596.491422923] [spawner_joint_state_broadcaster]: [94mLoaded joint_state_broadcaster[0m
2022-01-06T16:30:02.5165143Z 1: [ros2_control_node-6] [INFO] [1641486596.495135448] [controller_manager]: Configuring controller 'joint_state_broadcaster'
2022-01-06T16:30:02.5166626Z 1: [ros2 run controller_manager spawner joint_state_broadcaster-7] [INFO] [1641486596.510689949] [spawner_joint_state_broadcaster]: [92mConfigured and started [96mjoint_state_broadcaster[0m
2022-01-06T16:30:02.5168113Z 1: [ros2 run controller_manager spawner panda_joint_group_position_controller-8] [ros2run]: Process exited with failure 1
2022-01-06T16:30:02.5169390Z 1: [INFO] [ros2 run controller_manager spawner joint_state_broadcaster-7]: process has finished cleanly [pid 19640]
2022-01-06T16:30:02.5171013Z 1: [ERROR] [ros2 run controller_manager spawner panda_joint_group_position_controller-8]: process has died [pid 19650, exit code 1, cmd 'ros2 run controller_manager spawner panda_joint_group_position_controller'].
2022-01-06T16:30:02.5172040Z 1: Waiting for node...
2022-01-06T16:30:02.5172451Z 1: Waiting for node...
2022-01-06T16:30:02.5172856Z 1: Waiting for node...
2022-01-06T16:30:02.5173261Z 1: Waiting for node...
2022-01-06T16:30:02.5173654Z 1: Waiting for node...
2022-01-06T16:30:02.5174071Z 1: Waiting for node...
2022-01-06T16:30:02.5174475Z 1: Waiting for node...
2022-01-06T16:30:02.5174835Z 1: ok
2022-01-06T16:30:02.5175133Z 1: 
2022-01-06T16:30:02.5175990Z 1: ----------------------------------------------------------------------
2022-01-06T16:30:02.5176663Z 1: Ran 1 test in 3.013s
2022-01-06T16:30:02.5177005Z 1: 
2022-01-06T16:30:02.5177305Z 1: OK
2022-01-06T16:30:02.5178073Z 1: [INFO] [ros2_control_node-6]: sending signal 'SIGINT' to process[ros2_control_node-6]
2022-01-06T16:30:02.5179157Z 1: [INFO] [hybrid_planning_demo_node-5]: sending signal 'SIGINT' to process[hybrid_planning_demo_node-5]
2022-01-06T16:30:02.5180273Z 1: [INFO] [robot_state_publisher-4]: sending signal 'SIGINT' to process[robot_state_publisher-4]
2022-01-06T16:30:02.5181434Z 1: [INFO] [static_transform_publisher-2]: sending signal 'SIGINT' to process[static_transform_publisher-2]
2022-01-06T16:30:02.5182582Z 1: [INFO] [component_container_mt-1]: sending signal 'SIGINT' to process[component_container_mt-1]
2022-01-06T16:30:02.5183579Z 1: [ros2_control_node-6] [INFO] [1641486599.442942917] [rclcpp]: signal_handler(signum=2)
2022-01-06T16:30:02.5184496Z 1: [hybrid_planning_demo_node-5] [INFO] [1641486599.450237865] [rclcpp]: signal_handler(signum=2)
2022-01-06T16:30:02.5185598Z 1: [hybrid_planning_demo_node-5] [INFO] [1641486599.450512667] [test_hybrid_planning_client]: Initialize Planning Scene Monitor
2022-01-06T16:30:02.5186782Z 1: [robot_state_publisher-4] [INFO] [1641486599.451693475] [rclcpp]: signal_handler(signum=2)
2022-01-06T16:30:02.5187773Z 1: [static_transform_publisher-2] [INFO] [1641486599.453520887] [rclcpp]: signal_handler(signum=2)
2022-01-06T16:30:02.5188869Z 1: [hybrid_planning_demo_node-5] [INFO] [1641486599.453587487] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00281912 seconds
2022-01-06T16:30:02.5190038Z 1: [hybrid_planning_demo_node-5] [INFO] [1641486599.453628987] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
2022-01-06T16:30:02.5191061Z 1: [component_container_mt-1] [INFO] [1641486599.459429625] [rclcpp]: signal_handler(signum=2)
2022-01-06T16:30:02.5192281Z 1: [hybrid_planning_demo_node-5] [WARN] [1641486599.487360208] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
2022-01-06T16:30:02.5193842Z 1: [hybrid_planning_demo_node-5] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
2022-01-06T16:30:02.5195460Z 1: [hybrid_planning_demo_node-5]   what():  the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-rolling-rcl-4.0.0/src/rcl/guard_condition.c:67
2022-01-06T16:30:02.5196833Z 1: [hybrid_planning_demo_node-5] Stack trace (most recent call last) in thread 19674:
2022-01-06T16:30:02.5197792Z 1: [hybrid_planning_demo_node-5] #18   Object "", at 0xffffffffffffffff, in 
2022-01-06T16:30:02.5198874Z 1: [hybrid_planning_demo_node-5] #17   Object "/usr/lib/x86_64-linux-gnu/libc-2.31.so", at 0x7f9affa90292, in __clone
2022-01-06T16:30:02.5200102Z 1: [hybrid_planning_demo_node-5] #16   Object "/usr/lib/x86_64-linux-gnu/libpthread-2.31.so", at 0x7f9affdc5608, in start_thread
2022-01-06T16:30:02.5201360Z 1: [hybrid_planning_demo_node-5] #15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f9affc51de3, in 
2022-01-06T16:30:02.5202426Z 1: [INFO] [component_container_mt-1]: process has finished cleanly [pid 19608]
2022-01-06T16:30:02.5203400Z 1: [hybrid_planning_demo_node-5] #14 | Source "/usr/include/c++/9/thread", line 195, in operator()
2022-01-06T16:30:02.5204230Z 1: [hybrid_planning_demo_node-5]     |   194: 	void
2022-01-06T16:30:02.5204952Z 1: [hybrid_planning_demo_node-5]     | > 195: 	_M_run() { _M_func(); }
2022-01-06T16:30:02.5205654Z 1: [hybrid_planning_demo_node-5]     |   196:       };
2022-01-06T16:30:02.5206502Z 1: [hybrid_planning_demo_node-5]     | Source "/usr/include/c++/9/thread", line 251, in _M_invoke<0>
2022-01-06T16:30:02.5207361Z 1: [hybrid_planning_demo_node-5]     |   249: 	  using _Indices
2022-01-06T16:30:02.5208508Z 1: [hybrid_planning_demo_node-5]     |   250: 	    = typename _Build_index_tuple<tuple_size<_Tuple>::value>::__type;
2022-01-06T16:30:02.5209613Z 1: [hybrid_planning_demo_node-5]     | > 251: 	  return _M_invoke(_Indices());
[... 145 lines cut ...]
2022-01-06T16:30:02.5446045Z 1: [hybrid_planning_demo_node-5]         173: _GLIBCXX_NODISCARD inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
2022-01-06T16:30:02.5447313Z 1: [hybrid_planning_demo_node-5]       > 174: { return __p; }
2022-01-06T16:30:02.5448540Z 1: [hybrid_planning_demo_node-5]         175: _GLIBCXX_NODISCARD inline void* operator new[](std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
[... cut ...] mp/launch_params_pnhfu3f7'].
2022-01-06T16:30:02.5472220Z 1: [INFO] [robot_state_publisher-4]: process has finished cleanly [pid 19623]
2022-01-06T16:30:02.5473135Z 1: [INFO] [ros2_control_node-6]: process has finished cleanly [pid 19638]
2022-01-06T16:30:02.5473649Z 1: 
2022-01-06T16:30:02.5474268Z 1: ----------------------------------------------------------------------
2022-01-06T16:30:02.5474779Z 1: Ran 0 tests in 0.000s
2022-01-06T16:30:02.5475124Z 1: 
2022-01-06T16:30:02.5475409Z 1: OK
```

DavidMerzJr, 2022-06-02:
“I’ll work on this for WMD 2022”.

henningkayser, 2022-06-02:
@DavidMerzJr thank you! I think a fix would involve changing the settings to the correct plugin configuration.

closed 2023-03-10 by henningkayser