moveit/moveit2 #979: Error when trying to build Moveit2 from source

kevinshao7, 2022-01-09:
### Description
I'm encountering an error when trying to build Moveit2 from source. I'm new to ROS2 and I'd much rather install the binary debian packages (Yes, I remembered to source ROS2). However, it seems the Moveit2 tutorials require Moveit2 to be built from source. If anyone knows a easier way of installing Moveit2 tutorials please let me know.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Ros Binary Install
* (0.9.2-1focal.20211119.123452)

### Steps to reproduce
I was following [this](https://www.youtube.com/watch?v=k85fH0snRm8) tutorial.

### Expected behaviour
Missing packages automatically installed, allowing for successful colcon build

### Actual behaviour
Cannot locate rosdep definition for [moveit_visual_tools]

### Backtrace or Console output
~/ws_moveit2/src$ rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit2_tutorials: Cannot locate rosdep definition for [moveit_visual_tools]
Continuing to install resolvable dependencies...
executing command [sudo -H apt-get install -y ros-foxy-rviz-visual-tools]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-foxy-rviz-visual-tools
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-foxy-rviz-visual-tools] failed
  apt: Failed to detect successful installation of [ros-foxy-rviz-visual-tools]

Just for my own edification, why does Moveit2 use rviz visual tools? Isn't it a package for ROS and not ROS2?

tylerjw, 2022-01-09:
You are missing moveit_visual_tools in your workspace.  The tutorials don't have you building moveit2 itself from source.  It does have you building the tutorials and a couple of packages that are not yet released.  Here is the tutorial link I was following in that video: https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html

kevinshao7, 2022-01-09:
No, I followed all the steps, I just posted where things started to go wrong for me. A friend pointed out to me that I was working in the Ubuntu remote server GUI environment. I tried working with the Ubuntu terminal directly and it worked. I'm thinking it was a problem with the RAM available on my particular machine. In any case, thank you for the quick response.

closed 2022-01-09 by kevinshao7