moveit/moveit2 #990: Dependency on moveit is failing with rolling in our CI

BenjaminHug8, 2022-01-14:
### Description

Hello, we have a simulation example using `moveit` [here](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_universal_robot) but since we added a dependency on `moveit` in the [package.xml](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/package.xml#L25) some tests on our CI are failing as it cannot succeed to execute the command `apt-get install -y -qq ros-rolling-moveit`(example [here](https://github.com/cyberbotics/webots_ros2/runs/4815127148?check_suite_focus=true#step:3:5657) and we have also the same issue with `galactic testing`).

For `foxy` everything is fine so I am wondering if something change between `foxy` and `rolling`/`galactic` or is it something else?

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 20.04.3

Do not hesitate if I can give you more information!

Thank you in advance,
Benjamin

BenjaminHug8, 2022-01-17:
I listed the packages in [our CI](https://github.com/cyberbotics/webots_ros2/runs/4839126365?check_suite_focus=true#step:3:5263) and it seems to find `ros-foxy-moveit` but not `ros-galactic-moveit` and `ros-rolling-moveit`

BenjaminHug8, 2022-01-17:
I find out that the CI in the case of `galactic-testing` will search in this [source](http://packages.ros.org/ros2-testing/ubuntu/lists/ros-galactic-focal-amd64_focal_main_amd64_Packages) where there is no `ros-galactic-moveit` package.

Is something broken or should we target another source for our CI?

vatanaksoytezer, 2022-01-17:
Hello @BenjaminHug8 thanks for reporting. Seems moveit buildfarm builds are failing: https://build.ros2.org/job/Rbin_uF64__moveit__ubuntu_focal_amd64__binary/ because of a boost linking error in pilz: https://build.ros2.org/job/Gbin_uF64__pilz_industrial_motion_planner__ubuntu_focal_amd64__binary/15/console. We will work on a fix and try to send a patch release soon. Sorry for any inconvenience it might have caused. You can also find the buildfarm status of our packages in [moveit2 README](https://github.com/ros-planning/moveit2) in the very bottom of the page

vatanaksoytezer, 2022-01-21:
Releases has been cut. We just need them to get approved and synced: https://github.com/ros/rosdistro/pull/31876 https://github.com/ros/rosdistro/pull/31875

BenjaminHug8, 2022-01-24:
Perfect thank you!

closed 2022-01-24 by BenjaminHug8