moveit/moveit2 #1036: Unexpected behavior after calling getCurrentJointValues()

sabinevanderwerff, 2022-01-27:
Hello everyone,

I am facing issues, that seem to be related to a call to getCurrentJointValues(). I am planning a path to a certain point near a collision object, and then doing servo for the last part towards the collision object, after which is has to be attached to the robot. Between executing the path and starting the servo, I try to rotate the end-effector to 0, like so:
```
std::vector<double> joint_values = move_group_interface.getCurrentJointValues(); 
joint_values.at(5) = 0.0;
move_group_interface.setJointValueTarget(joint_values);
```
After this, I also add the collision object and the end-effector of the robot to the ACM, and then start servoing. When I do this sequence without calling getCurrentJointValues(), everything works fine. However, as soon as I call it, I get weird "jumps" at the start of the servoing. After that, the servo does proceed "normally" (as far as possible, it does execute the commands that are given). See the video for some visualisation. However, it seems like this is also related to a "Didn't received robot state (joint angles) with recent timestamp within 1 seconds." error, that I also only get when rotating the end-effector, but I am not _completely_ sure whether that is indeed the case. 

I have no clue how to deal with this, so any help is welcome. Thanks a lot in advance!
Sabine

### Your environment
* ROS Distro: ROS2 Foxy
* OS Version: Ubuntu  20.04
* Source or Binary build? Binary

### Steps to reproduce
Not sure if this gives the same behaviour for everyone, but:
- create a PoseStamped with a tf2 frame as frame_id and orientation as `quaternion.setRPY(0.0, 0.0, 0.0)`
- Plan a path using setPoseTarget;
- Execute
- Call getCurrentJointValues()
- Start servoing

### Expected behaviour
The servo motion starts at the same location as where the execution of the planned path ended.

### Actual behaviour
The robot jumps to a (random) location elsewhere

https://user-images.githubusercontent.com/54582035/151384215-b4009324-cc19-48c9-a202-aa39f5d54f0f.mp4

sabinevanderwerff, 2022-01-28:
Unfortunately, I have discovered that those jumps also happen without calling getCurrentJointValues(), although less frequent. I have not seen "Didn't received robot state (joint angles) with recent timestamp within 1 seconds." error without getCurrentJointValues(). I cannot yet pinpoint what then is the cause of this behavior, but I will close this issue for now, because getCurrentJointValues() is definitely not the cause.

gavanderhoorn, 2022-01-28:
Looks like an IK issue?

AndyZe, 2022-01-28:
I don't think you should close the issue, maybe just rename it.

It sounds like the problem is, Servo is not staying up-to-date with the joint state of the robot before you activate it. Is that what you think, as well?

How are you using Servo? Especially, how are you activating it? Are you using a service call or the C++ API?

AndyZe, 2022-01-28:
If you use the `setPaused(false);` function to activate Servo, here's the section of code where it spins in the background and listens to joint values. Adding a print statement should help you verify if it's listening to the robot state in the background.

https://github.com/ros-planning/moveit2/blob/5c405662f627f6d0a51de7c37edfc7c2a89a6bbd/moveit_ros/moveit_servo/src/servo_calcs.cpp#L364

sabinevanderwerff, 2022-01-28:
> It sounds like the problem is, Servo is not staying up-to-date with the joint state of the robot before you activate it. Is that what you think, as well?

Thank you both for your responses! I think this indeed describes what is happening. I am using a service call to activate servo, then I have a try and catch block for looking up certain transforms, and in a while loop I check whether the target was reached and if not, look up the transform again and publish a TwistStamped message. I do not think those transforms are the issue, this was already sometimes happening as well when I just used target coordinates. 

Below is the function that makes the call. Would there also be a way to check if the robot state is listened to when not using `setPaused(false)`?

```
bool MotionPlanner::unpauseServo(rclcpp::Client<std_srvs::srv::Trigger>::SharedPtr client_servo_unpause,
                                 std::shared_ptr<rclcpp::Node> node_servo)
{
    while (!client_servo_unpause->wait_for_service())
    {
        if (!rclcpp::ok())
        {
            RCLCPP_ERROR(rclcpp::get_logger("rclcpp"), "Interrupted while waiting for the service. Exiting.");
            return 0;
        }
        RCLCPP_INFO(rclcpp::get_logger("rclcpp"), "service not available, waiting again...");
    }

    auto trigger = std::make_shared<std_srvs::srv::Trigger::Request>();
    auto result = client_servo_unpause->async_send_request(trigger);
    // Wait for the result.
    if (rclcpp::spin_until_future_complete(node_servo, result) ==
        rclcpp::FutureReturnCode::SUCCESS)
    {
        RCLCPP_INFO(rclcpp::get_logger("rclcpp"), "succes");
        return true;
    }
    else
    {
        RCLCPP_ERROR(rclcpp::get_logger("rclcpp"), "Failed to call service");
        return false;
    }
}
```

AndyZe, 2022-01-28:
OK. It looks like the service call uses `setPaused(false);` internally so I'd still make the same suggestion about adding a print statement.

https://github.com/ros-planning/moveit2/blob/5c405662f627f6d0a51de7c37edfc7c2a89a6bbd/moveit_ros/moveit_servo/src/servo_node.cpp#L124

AndyZe, 2022-01-28:
Another question--  does the robot jump back to a spot where it was previously? Or does it jump to a whole new position?

If it jumps back to a previous position, then my earlier line of thinking is probably correct.

If it jumps to a new, random-seeming position, then I have no idea.

sabinevanderwerff, 2022-01-29:
On your last question, it seems to be a random position, the position that it jumps to in the video is for sure a position it has never been in previously. I will try to see if there is some pattern in it though, I am not really sure myself either. I currently do not have my code available so I will try that first thing monday. 

I also do want to verify whether Servo is listening to the robot state as well, just to make sure, but I have a binary installation, so I cannot add the print statement to servo_calcs.cpp. Do you know of another possibility to do so, or is it not possible to verify then?

AndyZe, 2022-01-29:
Is something else publishing /joint_states? Maybe a Gazebo simulation or something

sabinevanderwerff, 2022-01-31:
I can see from rqt_graph that only /joint_state_broadcaster publishes on /joint_states, subscribers are /robot_state_publisher, /servo_server_private and /move_group_private. /servo_server itself is not subscribed to /joint_states, but I doubt if that could explain things? I included a part of the rqt_graph of a cycle where this occurred.

I am not really sure whether this was clear before, but this behavior only happens every now and then, but I can't see what has changed in those cases. There might be ten runs with correct movement and then there might be one run with a joint jump the next time, which makes it hard to reproduce (or explain how to reproduce it).
[image]

AndyZe, 2022-01-31:
I'm sorry I'm not much more help. We have a project that does the same thing you're doing, and it doesn't have this issue.

sabinevanderwerff, 2022-02-01:
Your help is appreciated! If I find out what causes this, I will definitely put it here.

AndyZe, 2023-01-27:
If @juiceine or @BrezinaAdam could test #1904, I think it would fix the issue.

PR #1904 (2023-02-03): Servo: always keep publishing zero-velocity cmds. Eliminate possibility of joint jumps
### Description

Likely fixes #1036, #1885 

There are a lot of simplifications here:

- Always keep publishing zero-velocity commands. Previously, we kept track of the number of zero-velocity commands published to try to  prevent spammy messages. I think the extra complexity wasn't worth it.
- If there is not a valid command, don't jump back to the previously-commanded state. This was causing issue #1885. Instead, come to a gradual halt based on the robot's current position.

This was tested by the bug repo [... cut]
closed 2023-02-03 by AndyZe