moveit/moveit2 #1040: Build failure on Foxy/Windows in moveit_chomp_optimizer_adapter

lilustga, 2022-01-27:
### Description
Unresolved symbol error for collision_detection::CollisionDetectorAllocatorHybrid::NAME. I would have expected #936 to fix this but I'm still hitting the error with these changes. I have a theory that it's specific to my system due to other binaries and am investigating.

### Your environment
* ROS Distro: Foxy
* OS Version: Windows 11
* Source build
* If source, which branch: Foxy

### Steps to reproduce
* Install ROS on Windows
* clone moveit2/foxy
* 'vcs import < moveit2.repos'
* `IGNORED_PACKAGES=warehouse_ros_mongo`
* `colcon build --packages-skip-by-dep %IGNORED_PACKAGES%  --packages-skip %IGNORED_PACKAGES% --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo`

### Actual behaviour
Unresolved symbol error for collision_detection::CollisionDetectorAllocatorHybrid::NAME.

MikeWise2718, 2022-07-28:
I am hitting this now. Any resolution?

lilustga, 2022-08-03:
@MikeWise2718 From what I remember it was a red herring that was caused by a binary mismatch on my local system. It wasn't reproducible in build pipelines or other user's machines.

henningkayser, 2023-01-26:
I'm closing this since the cause has been a workspace issue. @lilustga please reopen if you think there is still something to be addressed.

closed 2023-01-26 by henningkayser

VinHub-Gitcent, 2023-10-17:
@MikeWise2718 Do you remember how you fixed it for you? I am facing the same issue now and I don't really have an idea on where to start here.

Haibara-z, 2023-12-04:
> I am hitting this now. Any resolution?

Do you fixed it? I am hitting this now. Any resolution?