moveit/moveit2 #1041: MoveItCpp vs MoveGroupInterface planning time differences

rrufat, 2022-01-27:
I have a problem regarding planning time differences between MoveGroupInterface and MoveItCpp. Since moveit_cpp is more oriented toward performance demanding cases I expected it to be faster than move_group. But I did some benchmarking (didn't use official package for it) for all available planners (ompl) and most of the time using movegroupinterface seems more reasonable as it is faster than moveit_cpp in planning time. I don't whether it is how it is supposed to be, there is something problematic with my code or there is some kind of optimization trick I am missing?

Environment
* ROS Distro: Galactic
* OS Version: PopOS 20.04
* Version: source[galactic branch]

I measured time using chrono to get exact execution time of the code and did comparison based on that. All programs were closed so they are equivalent in that sense.

Here is measurement part of the code for moveit_cpp:

        auto ts = std::chrono::high_resolution_clock::now();
        auto result = planning_components->plan(); 
        auto te = std::chrono::high_resolution_clock::now();
        auto plan_time = (std::chrono::duration<double>(te-ts)).count();
        
In moveit_cpp when I pass planning_params to the plan() function execution time gets a bit worse.

And for move_group:

        auto ts = std::chrono::high_resolution_clock::now();
        auto result = move_group->plan(target_plan);
        auto te = std::chrono::high_resolution_clock::now();
        auto plan_time = (std::chrono::duration<double>(te-ts)).count();

Here is some benchmark graphs I obtained

MoveGroupInterface
[image]
MoveItCpp
[image]

tylerjw, 2022-02-08:
Thank you for this issue.  This looks like you put a lot of work into this.  Would you mind putting the code you ran to produce these results into a repo so I could try to reproduce it and troubleshoot it?

tylerjw, 2022-02-08:
If you can't easily share a repo for us to reproduce this could you share your configs for moveitcpp and what you are sending to the motion planning request for move_group.  Our suspicion is that collision checking is configured differently.

rrufat, 2022-02-28:
moveit_cpp 
I was changing planner each time it I ran the benchmark to avoid lag due to sending whole planning_params

```
benchmark_node:
  ros__parameters:
    planning_scene_monitor_options:
      name: "planning_scene_monitor"
      robot_description: "robot_description"
      joint_state_topic: "/joint_states"
      attached_collision_object_topic: "/benchmark_node/planning_scene_monitor"
      publish_planning_scene_topic: "/benchmark_node/publish_planning_scene"
      monitored_planning_scene_topic: "/benchmark_node/monitored_planning_scene"
      wait_for_initial_state_timeout: 10.0
```

    planning_pipelines:
      #namespace: "moveit_cpp"  # optional, default is ~
      pipeline_names: ["ompl"]

    plan_request_params:
      planning_attempts: 10
      planning_pipeline: ompl
      max_velocity_scaling_factor: 0.1
      max_acceleration_scaling_factor: 0.1
      planner_id: "LazyPRMstar"
      planning_time: 3.0

function to calculate planning time [cpp]

```
std::vector<double> calculatePlanningTime(const moveit::core::JointModelGroup* joint_model_group_ptr, 
                                          moveit_cpp::MoveItCppPtr moveit_cpp_ptr,
                                          std::shared_ptr<moveit_cpp::PlanningComponent> planning_components,
                                          geometry_msgs::msg::PoseStamped& target_pose,
                                          std::string planner_id)
{
    double t_plan{0}, t_plan_min{0}, t_plan_max{0};
    double t_exec{0},t_exec_min{0},t_exec_max{0};
    double success_rate{0};

    moveit::planning_interface::PlanningComponent::PlanRequestParameters planning_params;  
    planning_params.planning_pipeline = "ompl";
    planning_params.planning_time = 3.0;
    planning_params.planning_attempts = 10;
    planning_params.planner_id = planner_id; // set planner

    geometry_msgs::msg::Pose target_pose2;
    target_pose2.orientation.w = 1.0;
    target_pose2.position.x = target_pose.pose.position.x;
    target_pose2.position.y = target_pose.pose.position.y;
    target_pose2.position.z = target_pose.pose.position.z;

    auto target_state = moveit_cpp_ptr->getCurrentState();
    (*target_state).setFromIK(joint_model_group_ptr,target_pose2);
    auto isValid = planning_components->setGoal(*target_state);

    if (isValid) 
    {
      std::vector<double> planner_time_set;
      std::vector<double> execution_time_set;
      std::vector<double> min_max;
      std::vector<int> success;
      int n {20}; // calculate multiple times for consistency

      for (int i=0;i<n;++i) {

        auto ts = std::chrono::high_resolution_clock::now();
        auto result = planning_components->plan(); //  planning_params
        auto te = std::chrono::high_resolution_clock::now();
[... 83 lines cut ...]
      }

      if (planner_time_set.size() != 0) 
      { // extract planning result metrics
        t_plan = std::accumulate(planner_time_set.begin(),planner_time_set.end(),0.0)/planner_time_set.size();
        t_plan_min = *(std::min_element(planner_time_set.begin(),planner_time_set.end()));
        t_plan_max = *(std::max_element(planner_time_set.begin(),planner_time_set.end()));
        t_exec = std::accumulate(execution_time_set.begin(),execution_time_set.end(),0.0)/execution_time_set.size();
        t_exec_min = *(std::min_element(execution_time_set.begin(),execution_time_set.end()));
        t_exec_max = *(std::max_element(execution_time_set.begin(),execution_time_set.end()));
      }

      success_rate = (int)std::accumulate(success.begin(),success.end(),0)/(double)success.size();
      writeToTerminal(planner_id,success_rate,t_plan,t_plan_min,t_plan_max,t_exec,t_exec_min,t_exec_max);
      return  std::vector<double> {t_plan,success_rate,t_plan_min,t_plan_max,t_exec,t_exec_min,t_exec_max};
    }

    else 
    {
      std::clog << "Target cannot be reached\n" ;
      return  std::vector<double> {t_plan,success_rate,t_plan_min,t_plan_max,t_exec,t_exec_min,t_exec_max};
    }

}
```

rrufat, 2022-03-23:
@tylerjw has there been any update on this bug?

henningkayser, 2022-06-02:
Rewarming this... @rrufat did you come to any conclusions? MoveGroup is running MoveItCpp internally, so there must be something different on how the planning pipelines or requests are being initialized. It's very safe to assume that MoveItCpp standalone is faster than MoveGroupInterface, since it's really the same, only that MGI has an extra ROS call for each request. My hot take would be that one of your setups is still running default settings while the other one has a custom ompl_planning.yaml settings. It's also possible that some defaults are different when using one or the other. Please verify the actual ROS parameters at runtime and check if there are any differences between your two instances.

henningkayser, 2023-03-10:
Closing due to inactivity. Also it was targeting Galactic which is eol.

closed 2023-03-10 by henningkayser