moveit/moveit2 #1043: [Foxy] launch-param-builder not released to yet

Briancbn, 2022-01-28:
### Description

Overview of your issue here.

`rosdep` failed
```bash
sudo apt-get install -y ros-foxy-launch-param-builder
```
The package haven't been sync from http://packages.ros.org/ros2-testing/ubuntu/pool/main/r/ to http://packages.ros.org/ros2/ubuntu/pool/main/r/. No big deal for now. Just would like to know when will be the next sync.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build - foxy branch latest

### Steps to reproduce
Download the repository into a ROS workspace and run rosdep
```
cd <path-to-worksapce>/
source /opt/ros/foxy/setup.bash
rosdep install --from-paths src --ignore-src -y
```

### Expected behaviour
All dependencies installed without issue

### Actual behaviour
Cannot find package.

### Backtrace or Console output
```
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y ros-foxy-launch-param-builder] failed
  apt: Failed to detect successful installation of [ros-foxy-launch-param-builder]
```

vatanaksoytezer, 2022-01-28:
Thanks for the issue @Briancbn. Release has been cut to rosdistro and merged: https://github.com/ros/rosdistro/pull/31883. We are just waiting for the next sync which I expect to be within a few weeks. Having said that, we don't use `launch-param-builder` in foxy branch of moveit2. Are you sure you are in the `foxy` branch of moveit2?

Briancbn, 2022-01-28:
Sorry, it was my bad. I used the master version for rosdep, but the foxy branch for the build. It was a bug on my side. Thanks for the help.

closed 2022-01-28 by Briancbn