moveit/moveit2 #1049: Cannot load panda model or any model to rviz2

Neathle, 2022-01-30:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Galactic
* OS Version: e.g. Ubuntu 20.04
* Binary, from "Getting Started"
* If binary, which release version? 
* If source, which branch?

### Steps to reproduce
Follow the tutorial "Getting Started" in the galactic version of the documentation and then the "Moveit Quickstart in Rviz" tutorial, when clicking the "ok" button nothing loads and the error appears.

### Expected behaviour
The model should load as well as the planning library 

### Actual behaviour
Nothing is loaded and an error is thrown in the console

### Backtrace or Console output
```
❯ ros2 launch moveit2_tutorials move_group.launch.py
[INFO] [launch]: All log files can be found below /home/neathle/.ros/log/2022-01-30-01-12-59-365223-neathle-pc-81453
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [81455]
[INFO] [static_transform_publisher-2]: process started with pid [81457]
[INFO] [robot_state_publisher-3]: process started with pid [81459]
[INFO] [move_group-4]: process started with pid [81461]
[INFO] [ros2_control_node-5]: process started with pid [81463]
[INFO] [mongo_wrapper_ros.py-6]: process started with pid [81465]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-7]: process started with pid [81467]
[INFO] [ros2 run controller_manager spawner panda_hand_controller-8]: process started with pid [81469]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-9]: process started with pid [81471]
[static_transform_publisher-2] [INFO] [1643501579.825026714] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'panda_link0'
[robot_state_publisher-3] Link panda_link1 had 1 children
[robot_state_publisher-3] Link panda_link2 had 1 children
[robot_state_publisher-3] Link panda_link3 had 1 children
[robot_state_publisher-3] Link panda_link4 had 1 children
[robot_state_publisher-3] Link panda_link5 had 1 children
[robot_state_publisher-3] Link panda_link6 had 1 children
[robot_state_publisher-3] Link panda_link7 had 1 children
[robot_state_publisher-3] Link panda_link8 had 1 children
[robot_state_publisher-3] Link panda_hand had 2 children
[robot_state_publisher-3] Link panda_leftfinger had 0 children
[robot_state_publisher-3] Link panda_rightfinger had 0 children
[robot_state_publisher-3] [INFO] [1643501579.828783872] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-3] [INFO] [1643501579.828845719] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-3] [INFO] [1643501579.828856009] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-3] [INFO] [1643501579.828863633] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-3] [INFO] [1643501579.828870647] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-3] [INFO] [1643501579.828877460] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-3] [INFO] [1643501579.828885315] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-3] [INFO] [1643501579.828892769] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-3] [INFO] [1643501579.828899602] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-3] [INFO] [1643501579.828906174] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-3] [INFO] [1643501579.828913328] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-3] [INFO] [1643501579.828920201] [robot_state_publisher]: got segment panda_rightfinger
[ros2_control_node-5] [INFO] [1643501579.838754010] [controller_manager]: update rate is 100 Hz
[move_group-4] [INFO] [1643501579.865611206] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00265439 seconds
[move_group-4] [INFO] [1643501579.865671380] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[move_group-4] Link panda_link1 had 1 children
[move_group-4] Link panda_link2 had 1 children
[move_group-4] Link panda_link3 had 1 children
[move_group-4] Link panda_link4 had 1 children
[move_group-4] Link panda_link5 had 1 children
[move_group-4] Link panda_link6 had 1 children
[move_group-4] Link panda_link7 had 1 children
[move_group-4] Link panda_link8 had 1 children
[move_group-4] Link panda_hand had 2 children
[... 106 lines cut ...]
[rviz2-1] [INFO] [1643501580.485872239] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ros2 run controller_manager spawner panda_arm_controller-7] [INFO] [1643501580.498445349] [spawner_panda_arm_controller]: Loaded panda_arm_controller
[ros2_control_node-5] [INFO] [1643501580.503549430] [controller_manager]: Configuring controller 'panda_arm_controller'
[ros2_control_node-5] [INFO] [1643501580.503672825] [panda_arm_controller]: Command interfaces are [position] and and state interfaces are [position velocity].
[ros2 run controller_manager spawner joint_state_broadcaster-9] [INFO] [1643501580.504163608] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[ros2_control_node-5] [INFO] [1643501580.504202352] [panda_arm_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-5] [INFO] [1643501580.504659412] [panda_arm_controller]: Action status changes will be monitored at 20.00 Hz.
[INFO] [ros2 run controller_manager spawner panda_hand_controller-8]: process has finished cleanly [pid 81469]
[ros2 run controller_manager spawner panda_arm_controller-7] [INFO] [1643501580.524340385] [spawner_panda_arm_controller]: Configured and started panda_arm_controller
[rviz2-1] [INFO] [1643501580.592121820] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-9]: process has finished cleanly [pid 81471]
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-1]          at line 253 in /opt/ros/galactic/include/class_loader/class_loader_core.hpp
[INFO] [ros2 run controller_manager spawner panda_arm_controller-7]: process has finished cleanly [pid 81467]
[rviz2-1] [ERROR] [1643501583.686525197] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1643501583.710021109] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00199648 seconds
[rviz2-1] [INFO] [1643501583.710054823] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-1] [WARN] [1643501583.959575558] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [INFO] [1643501584.076401882] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00143538 seconds
[rviz2-1] [INFO] [1643501584.076423724] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-1] [ERROR] [1643501584.083691370] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.panda_joint1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.panda_joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.
[rviz2-1] [INFO] [1643501584.085457140] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0017505 seconds
[rviz2-1] [INFO] [1643501584.085478270] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-1] [ERROR] [1643501584.090330672] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.panda_joint1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.panda_joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.
```

AndyZe, 2022-01-30:
Would you mind navigating to the moveit_resources folder and checking if you have this change:  https://github.com/ros-planning/moveit_resources/commit/f0a5d4ba2f7e7b4f8364ad75a5bea09b6150049e

Does it work if you have those changes?

Neathle, 2022-01-30:
> Would you mind navigating to the moveit_resources folder and checking if you have this change: [ros-planning/moveit_resources@f0a5d4b](https://github.com/ros-planning/moveit_resources/commit/f0a5d4ba2f7e7b4f8364ad75a5bea09b6150049e)
> 
> Does it work if you have those changes?

Those changes are present and it still fails

Neathle, 2022-01-30:
I also checked /parameter_events and the timeout is published with the correct type:
```
---
stamp:
  sec: 1643542734
  nanosec: 602135685
node: /moveit_simple_controller_manager
new_parameters: []
changed_parameters:
- name: robot_description_kinematics.panda_arm.kinematics_solver_timeout
  value:
    type: 3
    bool_value: false
    integer_value: 0
    double_value: 0.05
    string_value: ''
    byte_array_value: []
    bool_array_value: []
    integer_array_value: []
    double_array_value: []
    string_array_value: []
deleted_parameters: []
---
```

galou, 2022-02-21:
This is an issue with locales, adding `LC_NUMERIC=en_US.UTF-8` in front of your command should help (e.g. `LC_NUMERIC=en_US.UTF-8 ros2 launch moveit_task_constructor_demo demo.launch.py`). Cf. https://github.com/ros-visualization/rviz/issues/1151#issuecomment-345687074 for reference.

Neathle, 2022-02-21:
I will try it but it makes sense since I have a dual boot system and ubuntu is set on rtc time.

henningkayser, 2022-03-22:
@Neathle did setting the locale fix your issue? It's also documented as the first step in the ROS install instructions here https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html#set-locale. Please close this issue if you're good.

Neathle, 2022-03-23:
Sorry for not answering before, the problem is solved by setting the locale in UTC instead of RTC.

galou, 2022-03-23:
@Neathle you're confusing clock and locale here. The locale are regional settings, i.e., among others, the associated way to interpret numbers. I'm actually wondering how setting the BIOS time solved your issue.

Neathle, 2022-03-23:
I really don't know but there must be an association between the two, I haven't changed anything else.

henningkayser, 2022-04-12:
@Neathle is this resolved so that we can close this issue?

Neathle, 2022-04-18:
Yes no more errors thank you!

closed 2022-04-18 by stephanie-eng