moveit/moveit2 #1066: Build regression on ROS 2 Galactic

cottsay, 2022-02-07:
### Description

The `moveit` package regressed in ROS 2 Galactic about a month ago and still hasn't been resolved. I'm trying to get a sync of Galactic out the door, and this is currently the only regression.

I did some testing and #1027 fixes the issue. The sync is scheduled for Wednesday, so we have until then to either release this package or cherry-pick the fix into `moveit-release` to issue a patch release that fixes the build-blocking problem specifically.

Please let me know if there is anything I can do to help keep the package from regression.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu Focal
* Source or Binary build? Binary
* If binary, which release version? 2.3.3
* If source, which branch? N/A

https://build.ros2.org/view/Gbin_uF64/job/Gbin_uF64__pilz_industrial_motion_planner__ubuntu_focal_amd64__binary/

tylerjw, 2022-02-07:
@cottsay thank you for the ping.  I'll bloom right now.

tylerjw, 2022-02-07:
https://github.com/ros-planning/moveit2/pull/1067

tylerjw, 2022-02-07:
https://github.com/ros/rosdistro/pull/32078

cottsay, 2022-02-07:
Thanks for the quick action, here. I've merged the rosdistro release - we'll see how the builds shape up.

tylerjw, 2022-02-07:
Thank you.  The reason I hadn't cut it sooner was that I couldn't figure out how to run a pre-release test that would replicate the issue that broke the buildfarm so I didn't want to release again without figuring that out.

I'd still like a better pre-release test than what we have now.  For now we are using the one built into industrial_ci but I couldn't get it to reproduce the error found by the buildfarm.