moveit/moveit2 #1069: Attached Collision Object is not removed from the scene

AndyZe, 2022-02-08:
### Description

The attached Python script simply publishes an attached collision object on the `/attached_collision_object` topic. Then it removes the same object.

Drop the attached Python file into install/moveit2_tutorials/lib/moveit2_tutorials/. Make it executable with `chmod +x simple_collision_object_pub.py`. 

#### Steps to reproduce

Run the Move Group tutorial (https://moveit.picknik.ai/main/doc/examples/move_group_interface/move_group_interface_tutorial.html) then publish the attached collision object.

```
ros2 launch moveit2_tutorials move_group.launch.py

ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py

ros2 run moveit2_tutorials simple_collision_object_pub.py
```

Press Enter to publish the collision object.

Press Next in the RViz GUI to plan a motion. It fails because there is a collision object in the way. All good so far.

Now press Enter again to remove the attached collision object.

Press Next in the RViz GUI to plan another motion. It still fails. This is the bug.

### Your environment
ROS Rolling, built from source

[simple_collision_object_pub.py.txt](https://github.com/ros-planning/moveit2/files/8025552/simple_collision_object_pub.py.txt)

AndyZe, 2022-02-08:
This video shows the issue. Planning fails even after the ACO has been removed.

https://user-images.githubusercontent.com/11284393/153030252-3f4933ac-0c2b-4346-a0d0-849fe4ac6e13.mp4

AndyZe, 2022-02-08:
Looking at the code, I think I have a conceptual misunderstanding here -- specifying to REMOVE an ACO only detaches it. The object still exists in the world.

AndyZe, 2022-02-08:
Here's a Python script to remove then detach the ACO.
[detach_then_remove.py.txt](https://github.com/ros-planning/moveit2/files/8026129/detach_then_remove.py.txt)

v4hn, 2022-05-12:
To remove an attached object in one processing step (at least from the outside) you should actually send* a PlanningScene message with the robot state filled in to remove the attached object and the world object filled in to remove the (then detached) object from the world. This way you need to send only one message at least.

-- 
* apply scene changes via service, not via message topics! MoveIt2 should really discourage using these topics for one-step updates to the scene! They do not run synchronously and might even drop!